ChristopheJacquet/PiFmRds #21: Pi_Fm_Rds won't compile on Raspberry Pi 2

btonbadger, 2015-02-09:
Hi,

I'm unable to compile Pi_Fm_Rds on a Raspberry Pi 2.

There is no error message after the "make" command but the pi_fm_rds executable is not created even though other items are.

If I try and compile pi_fm_rds.c separately a number of errors occur:

root@pi-web-vpn:/home/pi/PiFmRds/src# make pi_fm_rds
gcc   pi_fm_rds.o   -o pi_fm_rds
pi_fm_rds.o: In function `terminate':
/home/pi/PiFmRds/src/pi_fm_rds.c:208: undefined reference to `fm_mpx_close'
/home/pi/PiFmRds/src/pi_fm_rds.c:209: undefined reference to `close_control_pipe'
pi_fm_rds.o: In function `tx':
/home/pi/PiFmRds/src/pi_fm_rds.c:424: undefined reference to `fm_mpx_open'
/home/pi/PiFmRds/src/pi_fm_rds.c:428: undefined reference to `set_rds_pi'
/home/pi/PiFmRds/src/pi_fm_rds.c:429: undefined reference to `set_rds_rt'
/home/pi/PiFmRds/src/pi_fm_rds.c:435: undefined reference to `set_rds_ps'
/home/pi/PiFmRds/src/pi_fm_rds.c:445: undefined reference to `open_control_pipe'
/home/pi/PiFmRds/src/pi_fm_rds.c:471: undefined reference to `poll_control_pipe'
/home/pi/PiFmRds/src/pi_fm_rds.c:499: undefined reference to `floorf'
/home/pi/PiFmRds/src/pi_fm_rds.c:488: undefined reference to `fm_mpx_get_samples'
/home/pi/PiFmRds/src/pi_fm_rds.c:465: undefined reference to `set_rds_ps'
/home/pi/PiFmRds/src/pi_fm_rds.c:461: undefined reference to `set_rds_ps'
collect2: ld returned 1 exit status
<builtin>: recipe for target 'pi_fm_rds' failed
make: *** [pi_fm_rds] Error 1

Any advice gratefully received.

BB

oncletom, 2015-02-11:
I suppose this is because of [that line](https://github.com/ChristopheJacquet/PiFmRds/blob/f8613175ae280de092bd31b924ad0ce38c9c8632/src/Makefile#L8) which explicitly states an `armv6` CPU whereas the Raspi2 is an `armv7`.

btonbadger, 2015-02-11:
I tried changing that but it still won't compile. Still no error message, it simply ignores pi__fm_rds.c during the make process.

oncletom, 2015-02-11:
When you type `uname -m`, what gets printed?

btweb, 2015-02-15:
Hi, I found the same thing. 

    # uname -m
    armv7l

I changed the architecture test to: ifeq ($(UNAME), armv7l) and compiled

    [root@alarmpi src]# make
    gcc -Wall -std=gnu99 -c -g -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math rds.c
    gcc -Wall -std=gnu99 -c -g -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math waveforms.c
    gcc -Wall -std=gnu99 -c -g -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math pi_fm_rds.c
    gcc -Wall -std=gnu99 -c -g -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math fm_mpx.c
    gcc -Wall -std=gnu99 -c -g -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math control_pipe.c
    gcc -o pi_fm_rds rds.o waveforms.o pi_fm_rds.o fm_mpx.o control_pipe.o -lm -lsndfile

So it compiles fine for me, however I don't seem to be getting any RF on GPIO4, nor any of the pins,  I went through all of them :)

It strikes me the Pi 2 is going to be much better at doing this, so am interested in helping find a solution to this.

I'm using Arch linux BTW with base-devel and libsndfile installed.

flashspys, 2015-02-21:
You have to change the DMA, PWM and CLK Base in `pi_fm_rds.c`. The Bases have changed from Raspberry Pi 1 to Raspberry Pi 2. Normally The base changed from 0x20… to 0x3f… but after changing the BASEs `pi_fm_rds` outputs: `Failed to reverse map phys addr 00000000` . gdb says the call comes from `pi_fm_rds.c:477` so maybe the address of the dma changed, too. Unfortunally the Peripherals documantation for the RPis 2 processor is not public. :(

DrSutherland, 2015-03-23:
So the "DMA, PWM and CLK Base..." changes do not fix this issue?

flashspys, 2015-03-23:
I think that not only the base adress changed. The address of DMA changed, too. But without Peripherals documantation its very hard to get it.

DrSutherland, 2015-04-08:
When you say "Peripherals documentation for the RPis 2 processor is not public", does that mean that the older RPi have their documentation published and they haven't released the RPi2 or they are not going to release it?

flashspys, 2015-04-08:
Just saw that the Raspberry Page was updated: http://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/README.md
Here spawned a new document for BCM2836 with peripheral documentation. I'll look over it this week.

DrSutherland, 2015-04-17:
Anything new with the peripheral documentation for the RPi2 ?

RamonSmit, 2015-04-28:
I'm also going to take a look at this, would be awesome to get this working on RPi 2

Bertsz, 2015-05-02:
Hi everybody, 
I know I'm a bit late to the party but I just got it working on RPi2. I can't really explain why, but I'll do my best.
So the first thing I did, was editing the pi_fm_rds.c. Basically, I just changed these lines from 0x20 -> 0x3F to look like this:
#define DMA_BASE		0x3F007000
#define PWM_BASE		0x3F20C000
#define CLK_BASE		0x3F101000
#define GPIO_BASE		0x3F200000

Compiling with the make-file didn't work, so I compiled it manually with these commands:
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math rds.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math waveforms.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math pi_fm_rds.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math fm_mpx.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math control_pipe.c
gcc -o pi_fm_rds rds.o waveforms.o pi_fm_rds.o fm_mpx.o control_pipe.o -lm -lsndfile

Notice the change from armv6 to armv7, because otherwise I would get this error "sorry, unimplemented: Thumb-1 hard-float VFP ABI".

./pi_fm_rds -audio sound.wav
Virtual memory mapped at 0x7693f000
ppm corr is 0.0000, divider is 1096.4912 (1096 + 2012*2^-12) [nominal 1096.4912]
Using audio file: sound.wav
Input: 228000 Hz, upsampling factor: 1.00
1 channel, monophonic operation.
Created low-pass FIR filter for audio channels, with cutoff at 12000.0 Hz
PI: 1234, PS: <Varying>
RT: "PiFmRds: live FM-RDS transmission from the RaspberryPi"
Starting to transmit on 107.9 MHz.

I expected a "Failed to reverse map phys addr 00000000" return, but this is even better.
Turns out the Pi2 is actually broadcasting, but only in mono, the audio is slowed down by around 30%, and occasionally it would only work after a few tries.
The Pi formerly used a media center with osmc on it had kodi still running. To fix the audio slowdown, I ran "systemctl stop mediacenter" to stop kodi.
However, when I wanted to start pi_fm_Rds, I would get this error: Failed to reverse map phys addr 0000000
Starting kodi again would make this error go away.
Hope this could be helpful in some way.

ChristopheJacquet, 2015-05-03:
Hi all, Sorry I don't have a Raspberry Pi 2 to test. I definitely need to get one.

Bertsz: With your change, is RDS working?

Bertsz, 2015-05-04:
These changes are necessary to make it work on the RPi2, but there's definitely more to it. Probably something with the mem_phys_to_virt function. At least that's where the programm stops working when I have my mediacenter disabled. As for the Radio Data System, unfortunately it didn't work. Maybe becasue of the slowdown I mentioned in the other post.

SandPox, 2015-06-05:
I got this https://github.com/markondej/fm_transmitter, it's working on Pi 2 but no stereo support and so much interferes

ChristopheJacquet, 2015-06-11:
@SandPox: thanks, I'll need to get my own RPi 2 to test, definitely. With your change, you say "no stereo support"... Is RDS working?

SandPox, 2015-06-12:
@ChristopheJacquet yes but it's very weird... sometime it's work sometime it's not

Bertsz, 2015-06-12:
@SandPox: If you really want stereo FM you could try installing OSCM on your RPi2, use PiFmRds, change the pi_fm_rds.c file as mentioned in one of my previous posts and then compile it. Running it with the -ppm parameter will compensate for the slowdown I expierienced and will make everthing work perfectly. Stereo and RDS data is transmitted properly with a value of about 125000. You can try it out with this command:
./pi_fm_rds -audio stereo_44100.wav -ppm 125000

SandPox, 2015-06-12:
@Bertsz oh.. but this need to fix because ~20 times try i got about 1 time working, ~0.5 it will make my Pi completely frozen and need hard reboot, the others is fail to reverse  map phys addr 00000000 or some random address so that's a dirty workaround

Bertsz, 2015-06-12:
@ SandPox Well, that's interesting, since this doesn't happen to me. But you're right there has to be a better way.

SandPox, 2015-06-12:
@Bertsz umm.. what's you mean it's doesnt happen to you? You got a fix that 100% chance it will work without the fail to reverse phys map? Gimme

ChristopheJacquet, 2015-08-31:
Now it compiles on the RPi 2, although it does not work.

ChristopheJacquet, 2015-09-05:
Thanks all. I have just committed a change that adds RPi 2 compatibility while retaining RPI 1 compatibility. It is based on [Richard Hirst's fix to ServoBlaster](https://www.raspberrypi.org/forums/viewtopic.php?p=699651#p699651).

closed 2015-09-05 by ChristopheJacquet