ChristopheJacquet/PiFmRds #24: Won't compile on Raspberry Pi B+

ishan-marikar, 2015-03-20:
Hey there. PiFmRDS doesn't seem to want to compile on my Raspberry Pi B+ and it keeps spitting out this error. Any ideas?

pi@raspberrypi ~/projects/PiFmRds-master/src $ make

gcc -Wall -std=gnu99 -c -g -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math pi_fm_rds.c
pi_fm_rds.c:100:21: fatal error: sndfile.h: No such file or directory
compilation terminated.
Makefile:31: recipe for target 'pi_fm_rds.o' failed
make: *** [pi_fm_rds.o] Error 1

peppi001, 2015-03-20:
Pi-FM-RDS, depends on the sndfile library. To install this library on Debian-like distributions, for instance Raspbian, run sudo apt-get install libsndfile1-dev.

ishan-marikar, 2015-03-26:
Oh, thanks c: I'll try it out as soon as possible.

closed 2015-03-26 by ishan-marikar