ChristopheJacquet/PiFmRds #28: MPX output to soundcard

peppi001, 2015-04-27:
Hi!
I woluld like to use PiFmRds with a 192kHz capable sound card. Is there any chance to do it?
I need only the MPX signal on the sound card output.

Thanks.
Peppi

ChristopheJacquet, 2015-04-27:
Hi,

PiFmRds generates baseband signal at 228 kHz. That was an easy choice
because the RDS carrier is at 57 kHz, so 228 = 4*57. If you downsample the
baseband signal, you can certainly get it to a 192 kHz sound card. You will
need to add the downsample code though.

peppi001, 2015-04-27:
Thanks for the quick answer.
The problem is: I'm only a user not a coder :(

ChristopheJacquet, 2015-04-30:
You can do an easy test without coding anything: use the `rds_wav` program provided with PiFmRds: it writes the baseband data to a WAV file, at 228 kHz. Then use an audio tool (e.g. Audacity) to resample it at 192 kHz, and play this on your sound card. You should get what you want.

Note that `rds_wav` is much more experimental, and more limited, than the main PiFmRds binary (I used it for debugging purposes, especially at the beginning of the project). But it would be very easy to bring it on a par with `pi_fm_rds`.

closed 2015-04-30 by ChristopheJacquet

peppi001, 2015-06-15:
Hi!
I tested the rds_wav program.
After I resample the generated wav file it works like a charm.
But I need the output to the soundcard, not to file.
Please if You have some free time modify the code to support MPX sound output too :)
Thanks.
Peppi