ChristopheJacquet/PiFmRds #29: Pure MPX input (192kHz) from external audio processor

Modulaattori, 2015-04-29:
Hi!

Is it possible to input 192kHz mono audio to use external processing (stereo + RDS)? This is how it's done in "real" FM transmitters, so it essentially would be simulating that operation.

ChristopheJacquet, 2015-04-30:
Yes, please see Issue #28 (duplicate).

closed 2015-04-30 by ChristopheJacquet

Modulaattori, 2015-05-01:
Hi! Nope, that not what I meant.. It's the other way around. The idea of mine was to modulate directly according to a input signal that is strictly 192kHz samplerate and mono. The signal is generated by a external audio processing software (Stereo Tool) which already contaings freshly chewed package for any receiver to receive (MPX).

But thanks for your amazing work anyway!

ChristopheJacquet, 2015-05-03:
Ok, that seems definitely doable too. You need to modify the main program so that instead of generating the RDS data (`rds.c`) and the FM MPX (`fm_mpx.c`), you directly feed your MPX samples in `tx()`. Indeed you can do away with most of PiFmRds' code.

Here's my advice: you almost only need to modify `pi_fm_rds.c`. Modify the sample rate to be directly 192 kHz instead of 228 kHz. And then directly feed your samples (line 495).

Hecker0549, 2024-05-29:
I also wanted to do this ,but i do not know how to code. Can you please code it for me?