ChristopheJacquet/PiFmRds #30: swap GPIO pin ?

frollic, 2015-05-05:
Don't know if this is possible at all, but I'd like to move the pin which is used
for transmitting to another GPIO pin, is it doable at all ? 

GPIO4/pin 7 is usually used by the RTC module addon (it uses pin 1-3-5-7-9), 
hence the need of moving the FM-pin ;)

thank you in advance

ChristopheJacquet, 2015-06-11:
I don't think it's doable. For what I recall, the PWM peripheral (at least in the configuration used by these FM transmitters) can only use this specific pin.

WangZhengmao, 2016-01-02:
I want to swap gpio pin too , PI 2 do not have hardware PWM so it uses software pwm which can run on any gpio pins , so I think is it possible to swap gpio pin but I don't know how to change the code.

pilovis, 2017-01-19:
https://www.raspberrypi.org/documentation/configuration/pin-configuration.md

zeno-bg, 2017-05-20:
So how do we change the pin? I guess we'd have to change this line:

#define GPFSEL0 (0x00/4)

From here https://pinout.xyz/pinout/gpclk I found that GPCLK is available on other pins as well (but I'm not sure if that is enough ...). So I tried changing that line to:
#define GPFSEL0            (0x00/6)

But still no luck..

pat45, 2018-03-22:
Hi !
I had the same problem, I wanted to swap from pin 7 (GPIO4)  to pin 29 (GPIO5) , it's the easier way to do, I think ? because these two GPIO are very similar

I succed, 

I changed the 3 registers : GPFSEL0 (7e20 0000) , and the adress (clock0)  of CM_GP0CTL (7c10 1070) et CM_GP0DIV (7c10 1074) ... to clock1 : 7c10 1078 and 7c10 107c. ........... and recompiled PiFmRds !
it's OK

how to do in the file 'pifmrds.c ' :
- lignes 241 et 351 : replace all the  '12'  by  '15'
- lignes 171 et 173 : replace  '70'  by  '78'
- ligne 174 : replace  '74'  by  ' 7c'

that's all ! , ... and now i'm going to try with other pin , just for the fun

have a good day

pat45, 2018-03-22:
Erratum :

Line 171 :  7e10......74   replace but  7e10.....7c      of course !  Sorry !

iztoogood, 2020-01-19:
Hello @pat45 
may you update your trick please ? because I found none of the lines you discribe. in the file named pi_fm_rds.c 

line 241 .... is blank
line 351 ...  fatal("Could not map memory.\n");
line 171 ... #define PWM_CTL            (0x00/4)
line 173 ... #define PWM_RNG1        (0x10/4)
lien 174 ... #define PWM_FIFO        (0x18/4)

when i look for var names :
#define GPFSEL0            (0x00/4) : (not eaqual to 7e20 0000 )
CM_GP0CTL : not found
#define CM_GP0DIV (0x7e101074) : (Okkk !!! \o/ )

pat45, 2020-01-19:
Hi !

 

Two years ago I had to solve the problem of swap the pin of the antenna on PiFmRds, because i want to connect a little screen 3,5 inches on my rasberry PI 3 ( on the GPIO connector)

 

It's a long time for me, and I don't remenber  exactly how i had succed to swap on pin 29, but I can research in the notes i wrote on the web, and in the doc of the precessor BCM I have at home, and tell you soon !

 

But since that day i found a better solution ,that I regularly use today : swap pin 38 by using the similitude beetween  pin 07 and pin 38 ( GPIO 04 and GPIO20 of the BCM ) : their same capacity to be used like a clock !). By the way  I have also changed the fréquency excursion and  the two limits ( low and high ) of fréquencies ( until 10/500M insteed of 76/108M ).

 

I give you the two files to do your tests by remplacing the original pi_fm_rds.c file in PiFmRds/src/ directory before compilation. But i remember '29' is a bit unstable with the mouse, I think ? 

 

 

 

For your information,I also do the same research for RPITX software. I was less lucky. I only succed to run pin 38 but not stop pin 07 , so i have two antennas in the same time !! FUNNY !! probably more powerfull transmission ?

 

 

Sorry for my 'bad english', i hope you'll be able to understood that i tried to explain. ?

 

 Patrick

> Message du 19/01/20 10:58
> De : "iztoogood" 
> A : "ChristopheJacquet/PiFmRds" 

> Copie à : "pat45" 
, "Mention"

iztoogood, 2020-01-22:
Hi @pat45 !
Thanks a lot for your quick answer...
It look like if you answered via mail... but attachements look to have not joined neither in the mail i received nor in the girhub conversation.

I'm not sure to have well understood...
not because of your english... but because of my knowledge in fm and gpio handling ... I hope i'll can see your files soon in the thread : 
https://github.com/ChristopheJacquet/PiFmRds/issues/30

StefPedro, 2020-04-25:
Hello, can you upload the files to Google Drive or something?
I somehow managed to burn out GPIO4 and i need the file to use another GPIO
Thanks!

jbdu94, 2020-05-18:
Hi all
Have you managed to use another GPIO? I have a lcd display and cannot use GPIO 4 as a consequence.
Please provide instructions so that o can test. By the way, have you put a filter between PI and antenna?

pat45, 2020-05-19:
( réponse en français, pour être plus clair ! ( sorry ! )

 

Bonjour,

 

J'ai eu le même problème que vous et beaucoup d'autres à cause de l'utilisation d'un petit écran tactile déjà installé sur le connecteur GPIO de mon raspberry Pi3.

 

Je n'ai pas trouvé d'aide précise sur le 'net' pour changer l'antenne utilisée par PiFmRds, alors j'ai étudié la documentation du processeur BCM 2835 trouvée sur le 'net'. Ce n'est pas éxactement le même processeur que celui du Raspberry Pi3, mais il fonctionne identiquement et a la même structure interne.

 

J'ai donc essayé plusieurs méthodes pour changer de broche sur le port GPIO du raspberry en changeant directement la programmation au niveau du processeur. J'aurais pu essayer d'utiliser un fichier  d'interface : /boot/dt-blob.bin   dans le rasbberry, comme le fait le logiciel RPITX, mais je n'ai pas exploré cette piste.

 

J'ai eu plusieurs résultats positifs, sur la broche 29 aussi par exemeple, mais pour moi la méthode la plus simple, la plus logique et la plus éficace est de permuter l'utilisation des 2 registres horloges  'clock' (  GPFSEL0  et  GPFSEL2 ) au niveau du processeur. Pour cela il suffit de modifier les lignes 186,241 et 242 du 'source' : pi_fm_rds.c , et ainsi activer la broche 38 à la place de la 7 sur le Raspberry !  .... c'est peut-être un peu radical, mais ça marche !

 

186:   #define GPFSEL2            (0x08/4)

241:   // GPIO20 needs to be ALT FUNC5 to output the clock

242:   gpio_reg[GPFSEL2] = (gpio_reg[GPFSEL2] & ~(7 << 0)) | (2 << 0);

 

J'ai aussi fait ce même essais avec RPITX, ça marche aussi mais pour le moment je ne parviens pas à désactiver la brcoche 7 ? , j'émets donc sur deux broches ( funny!)

 

Je joins à mon mail les documents sur le processeur BCM2835 et le fichier 'pi_fm_rds'  modifié. Ce fichier intégre d'ailleurs d'autres modifications amusantes ( élargissement de l'excursion en fréquence et de la bande d'émission )

 

 ( je n'ai pas essayé d'utilser des filtres en sortie vers l'antenne, pas utile pour l'utilisation que je fait de ce logiciel, et c'est totalement un autre sujet )

 

cordialement

Patrick

 

f> Message du 18/05/20 21:39

> De : "jbdu94" 
> A : "ChristopheJacquet/PiFmRds" 

> Copie à : "pat45" 
, "Mention"

jbdu94, 2020-05-19:
Merci Patrick
Mais du coup, tu peux utiliser ton LCD et rpitx en même temps?
Parce que si GPIO 4 est allumé (si on peut dire, s’il émet quoi), ça pose pas de problème avec l’écran?

For all, I just asked in French if he managed to keep the LCD working while having rpitx transmitting on both GPIO he mentioned (including the GPIO 4 (pin 7) on which the LCD also is.

pat45, 2020-05-19:
Il a du m'arriver d'être dans cette configuration sur mon Raspberry équipé d'un LCD. De mémoire ça ne pose pas de problèmes :  ni à RPITX , ni à l'écran, ni à pifmrds quand on reste sur la broche 7   ??  mais pour moi ce n'est pas satisfaisant.

Je pense qu'avec encore un peu de 'matière grise' on devrait aboutir au même résultat qu'avec pifmrds.

 

Il faut dire que j'utile également un autre Raspberry Pi3  'de bureau'  sur lequel je fais mes développements informatiques (donc sans écran LCD, et que je rencontre rarement cette situation ) . Il faut dire également que je n'utilise quasiment plus RPITX du fait de sa médiocre qualité sonore comparé à pifmrds. Que dire en AM où c'est une véritable catastrophe !!!

 

Pour en revenir à pifmrds, je voudrais signaler que la modification N°3 ( l'ouverture de la bande d'émission ) n'a pas était testée totalemnt et  n'a pour le moment qu'un interêt limité. Mais elle permet d'émettre un peu  au delà des 108 Mhz , de sortir de la bande et d'éliminer les perturbation des autres émetteurs FM, ... et de ne géner personne ( en faisant attention toutefois aux harmoniques) .  Bien sûr pour cela il fait disposer en réception d'un bon vieux tuner FM analogique qui monte lui aussi toujours un peu au delà des 108 Mhz ( on peut aussi l'aider un peu si nécessaire ).  Aucun intérêt avec les tuners numériques modernes évidemment qui refusent obstinemment de sortir de leur bande, sniff !

 

 

Salutations à tous

Patrick

> Message du 19/05/20 08:38
> De : "jbdu94" 
> A : "ChristopheJacquet/PiFmRds" 

> Copie à : "pat45" 
, "Mention"