ChristopheJacquet/PiFmRds #31: PiFmRDS on Raspberry Pi B+ Compile Error VFP register

jpmendez79, 2015-06-06:
gcc -o pi_fm_rds rds.o waveforms.o pi_fm_rds.o fm_mpx.o control_pipe.o -lm -lsndfile
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: error: rds.o uses VFP register arguments, pi_fm_rds does not
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: failed to merge target specific data of file rds.o
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: error: waveforms.o uses VFP register arguments, pi_fm_rds does not
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: failed to merge target specific data of file waveforms.o
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: error: pi_fm_rds.o uses VFP register arguments, pi_fm_rds does not
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: failed to merge target specific data of file pi_fm_rds.o
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: error: fm_mpx.o uses VFP register arguments, pi_fm_rds does not
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: failed to merge target specific data of file fm_mpx.o
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: error: control_pipe.o uses VFP register arguments, pi_fm_rds does not
    /usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/../../../../armv6j-softfp-linux-gnueabi/bin/ld: failed to merge target specific data of file control_pipe.o
    collect2: error: ld returned 1 exit status
    Makefile:11: recipe for target 'app' failed
    make: *** [app] Error 1

Compile fails with this error.  If it helps, here is the gcc version and specs output:
   
    localhost src # gcc -v --dumpspecs 
    Using built-in specs.
    COLLECT_GCC=/usr/armv6j-softfp-linux-gnueabi/gcc-bin/4.8.4/gcc
    COLLECT_LTO_WRAPPER=/usr/libexec/gcc/armv6j-softfp-linux-gnueabi/4.8.4/lto-wrapper
    gcc: error: unrecognized command line option '--dumpspecs'
    Target: armv6j-softfp-linux-gnueabi
    Configured with: /var/tmp/portage/sys-devel/gcc-4.8.4/work/gcc-4.8.4/configure --host=armv6j-softfp-linux-gnueabi --build=armv6j-softfp-linux-gnueabi --prefix=/usr --bindir=/usr/armv6j-softfp-linux-gnueabi/gcc-bin/4.8.4 --includedir=/usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/include --datadir=/usr/share/gcc-data/armv6j-softfp-linux-gnueabi/4.8.4 --mandir=/usr/share/gcc-data/armv6j-softfp-linux-gnueabi/4.8.4/man --infodir=/usr/share/gcc-data/armv6j-softfp-linux-gnueabi/4.8.4/info --with-gxx-include-dir=/usr/lib/gcc/armv6j-softfp-linux-gnueabi/4.8.4/include/g++-v4 --with-python-dir=/share/gcc-data/armv6j-softfp-linux-gnueabi/4.8.4/python --enable-languages=c,c++,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo 4.8.4 p1.5, pie-0.6.1' --enable-libstdcxx-time --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --disable-multilib --disable-altivec --disable-fixed-point --with-float=softfp --with-arch=armv6j --disable-libgcj --enable-libgomp --disable-libmudflap --disable-libssp --enable-lto --without-cloog --disable-libsanitizer
    Thread model: posix
    gcc version 4.8.4 (Gentoo 4.8.4 p1.5, pie-0.6.1)

SandPox, 2015-06-07:
try to remove -mfpu=vfp in Makefile

jpmendez79, 2015-06-10:
After testing, I found a config that works.

Removing -mfpu=vfp does not solve, same error.
Removing -mfloat-abi=hard compiles cleanly and produces an fm signal
Changing to -mfloat-abi=soft compiles cleanly but produces a triple echo sound effect to broadcast audio

And finaly changing to -mfloat-abi=softfp compiles cleanly and produces a decent fm signal.

Looks like the error was related to -mfloat-abi instead of -mfpu.  The relevant line from my gcc version information that gave me the "softfp" idea was

    Target: armv6j-softfp-linux-gnueabi

Thanks for the tip to start changing compiler flags.

closed 2015-06-10 by jpmendez79