ChristopheJacquet/PiFmRds #33: Generation time of the RDS messages

Fernuts92, 2015-06-10:
This is not really an issue, it is more a doubt about the program but I think that it is an interesting question and I couldn't find a better way of asking it. 

I'm using your program in my MSc Thesis. I'm basically using the RDS as a control channel in a wireless mesh sensor network. The thing is, the default rate at which the RDS messages are being  generated in this program is too slow for my needs. I was able to increase that rate by changing the usleep period (line 475 of the pi_fm_rds.c file). Using a usleep of 1 in that cycle, I was able to increase the message generation rate to 1 RDS message from type 0A every 150 ms approximately (I have also done some modifications to only generate 0A messages). This increased rate is achieved at the cost of range and error rate. 

What I wanted to know is why we need this sleep and why it has to bee so long if we want the RDS transmission to be reliable and have a large range. If I got it right, you use this sleep between getting the samples but I would like to know more technical details. Moreover, according to my research, the physical limit for the bitrate of the Radio data System is 1187,5 bits per second. (as stated in this article http://www.iaeng.org/publication/WCECS2013/WCECS2013_pp431-436.pdf). With your program, the bit rates that we can achieve are much lower. Is this due to a limitation of the PWM generator of the Raspberry Pi?

To summarize, I just wanted to have an explanation of why the RDS generation has to be so slow in order to be reliable. 

Thank you very much in advance. This is a limitation of the RDS that I would like to address in my MSc Thesis.

Best regards.

ChristopheJacquet, 2015-06-10:
Hi,

When you say "the physical limit for the bitrate of the Radio data System
is 1187.5 bits per second" it's not entirely exact. 1187.5 bit/s is not an
"upper bound". The standard says that the bitrate must be *precisely*
1187.5 bit/s. The maximum deviation from this is ± 0.125 bit/s. See the
standard (ftp://ftp.rds.org.uk/pub/acrobat/rbds1998.pdf) page 8.

So PiFmRds generates the bitstream as close to 1187.5 bit/s as possible.
Clocking of samples (and hence bits) is done by the DMA. The timings are
located around line 381. As for line 475, each iteration of the loop. The
usleep() is not critical, and it is not directly related to the bitrate: it
is here to render the program idle while the DMA does the hard work. Every
once in a while, the program goes out of sleep and replenishes the DMA
buffers.

Fernuts92, 2015-06-10:
Hi Christophe,

Thank you for your quick response. That makes sense, I hadn't perceived it that way, I should have read the standard more carefully.

Going back to your program, based on my empirical knowledge, when modifying the sleep time of the program, the reception of the RDS messages was significantly affected. However, I have done some adaptations to your program and it's possible that I have screwed something up. I tried to split your tx function into an init function and a sender on demand function, i.e., I tried to modify the program in such a manner that, by calling a function, I would send a single RDS message. To do so, I assumed, by inspection of your for(;;) cycle, that every 1024 iterations a new RDS message was sent. Maybe this wasn't a good approach. It seemed to work alright at first, but then I realized that the RDS range and reliability was gone. I had errors around 5% with the transmitter and receptor at less than a meter from each other.

My problem is the following, I'm using the PI field of the RDS messages for my control protocol. Therefore, I need to change the PI field frequently, the faster the better. The PI field is set by an external program using shared memory, when the shared memory content is changed, the PI field is set in your program. If I used your program as it is, with an extra verification to check if changed memory changed, and if so, set the PI to the new value, where do you think that I should place these modifications? In my opinion the best place would be inside one those if's, when varying PS, maybe inside the if(count==512) (line 459). What do you think of this approach? 

Thank you very much for your help!

ChristopheJacquet, 2015-06-11:
Sample are sent by the DMA at 228k samples per second. 4 samples correspond
to a period of the carrier wave (carrier at 57 kHz). 48 periods correspond
to 1 bit (1187.5 bit/s). The for(;;) loop wakes up every 5 ms and
replenishes the DMA buffers. It has its own buffers, but when it calls
fm_mpx_get_samples to generate more baseband samples, it generates
DATA_SIZE (= 5000) of them at once.

In the RDS standard, the bitstream is transmitted continuously, so there is
no such thing as a "message" sent at one point in time. If you're using an
RDS demodulation chip, I strongly advise you to keep it this way, or
include a "guard" period of several tens of bits at the beginning of each
message for the receiver to sync. If you play with the sleep duration, I
suppose you can cause the DMA buffers to be depleted, hence an interruption
of the data stream.

My demo program changes the station name every 512 iterations of the
for(;;) loop. This is just for demo purposes. By playing with the size of
buffers, you can make it so that fm_mpx_get_samples is called much more
often, and that every RDS group contains different data.

I'm wondering: you say that you inject your specific data into the PI
field. Therefore it's no longer standard RDS. If you need an higher data
rate, nothing prevents you from using all the 4 blocks of each group, not
just block A (PI)... It will be even less standard, but I guess you don't
care (?).

Fernuts92, 2015-06-12:
Well, if I got it right, I think that I will continue with the same problem but now I have a justification for it at least. My problem is not the bit rate per se, want I wanted was to , say every 20 ms, change the data that I'm broadcasting. However, that doesn't seem possible since the samples for a RDS group are created all at once, although they are generated as a data stream across time. 

If it was possible to define the data I want to transmit in each block at different times, that would do what I want. For example, at t=0 I say to transmit "2" in block 0. While block 0 is generated, I say I want to transmit 3 in block 1 and when block 0 was completely sent (at t=21.5 ms +/-) the generator would start do broadcast the block 1 with the data I defined during the generation of block 0. In practice, I think this isn't possible because the generator already has the samples and the new data that I defined would only be transmitted when the program replenishes the DMA buffers. Is that correct? 

If this is true, maybe there is a way of playing with the buffers in such a manner that, in addition to each RDS group containing different data, each block would also contain different data, set at different time instants. Instead of getting a RDS group at once and generating samples for it, maybe I could do the same but with a single RDS block?

closed 2016-11-26 by ChristopheJacquet