ChristopheJacquet/PiFmRds #34: Broadcast Computer's Audio

jamesmstone, 2015-06-16:
Hi 
I was wondering if it's possible to broadcast the computer's audio over fm radio  / how would it be done? 
by computer's audio I mean what would be playing out of the speakers?
Thanks heaps
James

SandPox, 2015-06-18:
no, only can if you setup a stream server on your computer and grab sound via LAN like 
sox -t mp3 http://www.linuxvoice.com/episodes/lv_s02e01.mp3 -t wav -  | sudo ./pi_fm_rds -audio -

jamesmstone, 2015-06-18:
Ahh ok, thanks for the help :)

ChristopheJacquet, 2015-09-04:
Closing, outside of the scope of PiFmRds, which is for experimentation only, and not intended as a broadcast *tool*.

closed 2015-09-04 by ChristopheJacquet

allegfede, 2015-09-29:
by the way. remember to add sox mp3 support with:
sudo apt-get install libsox-fmt-mp3