ChristopheJacquet/PiFmRds #37: Program stops after a few hours

danielmoore123, 2015-08-27:
I know this has come up before, but it still does not seem to be fixed.... basically, when running this, it will turn itself off after a few hours and stop sending RDS and audio, it will however keep the signal occupied. Is there any reason why it stops sending audio and RDS after a few hours? Also, is it possible for this to work 24/7 (for as long as the Sox incoming audio feed lasts)? 
The pi does not shut down and after a few hours it behaves like it would if you were to terminate by pressing CTRL + C. I simply restart it by pressing UP then ENTER but I don't wish to run it like this.

Hatagashira, 2015-09-14:
Are you sure this isn't just bug #16 ?

danielmoore123, 2015-10-12:
I'm sure it's not, the program no longer crashes the Pi or falls over itself it just cleanly stops itself.... If I press UP then ENTER and re-insert the code it goes again for the same length of time, any chance it can run until I stop it as opposed to keep having to restart it?

WangZhengmao, 2016-01-12:
I found this too, and I am sure it isn't bug #16 ,if you use this to broadcast musics in your play list, I foud a method to deal with this .

#This is my code

mympg123()
{
mpg123 -y --stereo -r 44100 --buffer 4096 --no-icy-meta -q -s "$filename"| sox -v $vol -t raw -b 16 -e  signed -c 2 -r 44100 - -t wav - equalizer 11800 0.7q 20 2>/dev/zero&

echo "RT $songname" >rds_ctl & #if PI_fm_rds stoped,this command cannot be finished .
BG_PID=$! #so if the pid still exists after 1 second,that is to say PI_fm_rds stoped.
sleep 1
weatherexit=`kill $BG_PID 2>&1`
if [ -z "$weatherexit" ] ;then
echo An error occured,restarting >&2
exit

fi

danielmoore123, 2016-01-17:
This is my current code: 

sudo sox -t mp3 http://10.43.87.11:8000/live.m3u -t wav -r 44100 -c 2 - | sudo ./pi_fm_rds -freq 107.0 -ps 'East 105' -ctl rds_ctl -rt 'The biggest hits for the east coast' -audio -

How can I get this to loop/pick itself up when it dies? I tried your code but it makes very little sense to me..... How can I get the URL to be the master input? It's a LAN URL which is always up so shouldn't have an issue there, the RT does not change but when it does it is done remotely on another user account on the Pi. What's important is that the audio does not stop.
Cheers

danielmoore123, 2016-01-19:
@WangZhengmao your loop only seems to loop the termination that the program automatically performs after a few hours with my test using sox to pipe mp3 audio into the program, so for example when CTRL + C is pressed to terminate it loops termination and a force reboot is needed to kick it out, good theory though. Would a loop like this keep a live feed from an Aux cable alive? I don't know how to pipe aux in to the program but would like to instead of piping in an internet stream.

Is there a way this can be achieved?

danielmoore123, 2016-02-04:
I now pipe in via USB Sound card, but now, this happens:

overrun!!! (at least 0.503 ms long)
overrun!!! (at least 0.400 ms long)
overrun!!! (at least 0.614 ms long)
overrun!!! (at least 4.543 ms long)
overrun!!! (at least 1.988 ms long)
overrun!!! (at least 0.070 ms long)
Could not rewind in audio file, terminating
Terminating: cleanly deactivated the DMA engine and killed the carrier.

Niall7459, 2017-11-11:
That is a problem with either aplay or arecord or whatever command you're Using. I have the same problem so I have a script that kills the programme every 2 hours then restart it