ChristopheJacquet/PiFmRds #38: Raspberry pi2 problem

clotaireFrance, 2015-09-05:
Hello 

I ve got a problem

following your tutorial, then:
root@raspberrypi2:/home/pi/PiFmRds/src# ./pi_fm_rds
I get , on my raspi 2:
Can't open device file: mbox
Try creating a device file with: sudo mknod mbox c 100 0
just discovering git hub
 too difficult for me to understand the problem

TY

//--------------------------------------------------------------
bonjour,
J'ai un soucis lors du lancement du programme

en suivant tes explications puis :
root@raspberrypi2:/home/pi/PiFmRds/src# ./pi_fm_rds
j obtiens:
Can't open device file: mbox
Try creating a device file with: sudo mknod mbox c 100 0

ChristopheJacquet, 2015-09-05:
You need to have a recent Linux kernel (possibly patched, I need to check this) because now PiFmRds needs the rpi-mailbox driver. I use a Raspbian from early August and it works out of the box.

SandPox, 2015-09-06:
I have this problem too, running Debian 9/testing (real debian, not raspbian) and latest 4.1 kernel

clotaireFrance, 2015-09-06:
What a fast reply!
'Just tried it on my other RspbB+, it works fine,   so funny
Although I'm a mac gyver, this kind of code ilt's too difficult for me
Can't help you...

ChristopheJacquet, 2015-09-06:
@SandPox: No need to run mknod, pi_fm_rds does this for you. But if it fails, it means your kernel lacks the required driver. I'm will update the message and doc when I figure out the specifics.

It works fine on my Raspbian with Linux 3.18.11 (but the kernel may have been patched with RPi-specific patches by the Raspbian team).

ChristopheJacquet, 2015-09-06:
@SandPox: may you try to change the code here in pi_fm_rds.c?

	if (mknod(MBFILE, S_IFCHR|0600, makedev(100, 0)) < 0)

Change 100 into 249.

May you also send me the output of ls -l /dev/vcio?

Thanks.

clotaireFrance, 2015-09-06:
I get

crw-rw---T 1 root video 249, 0 janv.  1  1970 /dev/vcio

clotaireFrance, 2015-09-06:
sorry i will change the value and retry

one minute

ChristopheJacquet, 2015-09-06:
You will also need to change MAJOR_NUM 100 to 249 in mailbox.h.

clotaireFrance, 2015-09-06:
i do this in 5 minutes
( the soft is able to run with the first change in pi_fm_rds.c)

ChristopheJacquet, 2015-09-06:
Ah great :) Don't bother, I checked again, the change in mailbox.h will have no effect.

clotaireFrance, 2015-09-06:
so:
first change on pi_fm_rds.c the soft run
second change on mailbox.h  the rasp is blocked

i get:

Allocating physical memory: size = 3403776     ioctl_set_msg failed:-1
mem_ref = 3403776     ioctl_set_msg failed:-1
bus_addr = 33f000     virt_addr = 0x76813000

ChristopheJacquet, 2015-09-06:
Ah interesting, I did not expect that, thanks :)

clotaireFrance, 2015-09-06:
My test
I reset  MAJOR_NUM at  value 100 in mailbox.h 
the soft run,  i can hear stereo_44100.wav at good frequency with RDS on my radio

but it seems to be in mono

ChristopheJacquet, 2015-09-06:
That's weird, stereo should not be affected by the issue we're discussing here. I'll do some tests later.

clotaireFrance, 2015-09-06:
it's emmiting in stereo but the sound is in mono

anyway
good job,  T Y.

ChristopheJacquet, 2015-09-06:
I pushed a new version: hopefully it fixes the problems you reported with the device file. It tries to use /dev/vcio, then create a device 249 (new allocation), then create a file with major 100 (old allocation).

SandPox, 2015-09-07:
wow, new commit working... thanks

clotaireFrance, 2015-09-07:
Hello,

Now Working fine for Raspberry Pi 2 !  Thanks!

closed 2015-09-07 by ChristopheJacquet