ChristopheJacquet/PiFmRds #39: Program not compiled on Raspberry 2

hlc-mac, 2015-09-10:
Hi,
After a clone of repository, I have this error when i compiled :
pi@raspberrypi:~/PiFmRds/src$ make clean
rm *.o
rm: cannot remove `*.o': No such file or directory
Makefile:51: recipe for target 'clean' failed
make: *** [clean] Error 1
pi@raspberrypi::~/PiFmRds/src$ make
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 rds.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 waveforms.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 pi_fm_rds.c
pi_fm_rds.c:105:21: fatal error: sndfile.h: No such file or directory
compilation terminated.
Makefile:42: recipe for target 'pi_fm_rds.o' failed
make: *** [pi_fm_rds.o] Error 1

ChristopheJacquet, 2015-09-10:
Hi, You need to install libsndfile (see the README).

farmazon3000, 2015-12-14:
I had exactly the same problem and what @ChristopheJacquet says solves it, so this issue can be closed.

closed 2016-01-17 by ChristopheJacquet

dhana0809, 2016-07-14:
Important. The binaries compiled for the Raspberry Pi 1 are not compatible with the Raspberry Pi 2, and conversely. Always re-compile when switching models, so do not skip the make clean step in the instructions below!

Clone the source repository and run 
git clone https://github.com/ChristopheJacquet/PiFmRds.git
cd PiFmRds/src
make
make clean
make