ChristopheJacquet/PiFmRds #44: Tip for streaming any content

dehy, 2015-10-17:
Hi there, thank you for your wonderful software!

In your examples, you talk about sox to stream non-wav data to PiFmRds. Unfortunately, some protocols (like HTTP Live Streaming, or extracting audio from video) are not supported with sox.

To play nearly everything to FM, you can use LibAV (http://libav.org) or FFMPEG (http://ffmpeg.org but not availble out-of-the-box on Raspbian) but it has some delay before outputing the result on stdout, resulting on PiFmRds stopping because it can't immediately read stdin.

Here is a workaround that may be useful for some fellows. Launch on 2 separate terminals or background the first process :

    $ sudo avconv -re -i http://url.of.the/file-or-stream -vn -sn -f mp3 udp://127.0.0.1:8080
    $ nc -l -u 8080 | sox -t mp3 - -t wav - | sudo ./pi_fm_rds -audio -

danielmoore123, 2015-11-07:
Hi dehy, thanks for the tip, the issue I have is the program cleanly stops working after 2-3 hours, if I re-run the command (without rebooting) it works fine for around the same length of time.
I am using sox like so:

sox -t mp3 http://[stream.url]:8000 -t wav - | sudo ./pi_fm_rds -freq 96.0 -pi FFFF -ps PIRDS -rt 'PI RDS DEMO' -ctl pifm -audio -

What could I be doing wrong? Something is causing it to stop but obviously I would like to fix it so it keeps running until I press stop myself. The above code works for me but quits, is this the stream or the software? Either or, could you maybe help me to fix this please?
Thanks in advance.

Blenderpics, 2016-03-04:
The easiest way to basicly play anything you want to, is by using audio loopdevices.
Simply load snd-aloop via modprobe (or just add it to /etc/modules), and you are ready to go.
use a terminal to start PiFm:
`sudo arecord -fS16_LE -r 44100 -D hw:Loopback,1,0 -c 2 -  | sudo ./pi_fm_rds -audio -`

then you are free to play whatever you want with whatever program you want, as long as it allows you to specify the audio device you want to use to output to.
for example:
`sudo aplay -D hw:Loopback,0,0 "FILE"`

if you want to, you can even define the loopdevice as your standard output device, by changing the .asoundrc in your home directory to
`# .asoundrc
pcm.!default { type plug slave.pcm "hw:Loopback,0,0" }`

That way you can even use VLC within X to transmit your audio.

nickpiggott, 2016-04-10:
I'm piping like this to keep audio quality as high as possible:

`avconv -i $1 -f ogg - | sox -t ogg - -t wav - | sudo pi_fm_rds -audio - -freq $2 -pi $3 -ps $4 -rt $5`

I did try modifying the source code of fm_mpx.c to handle the source differently, but the use of sf_read_float makes that refactoring more complicated.

13321998692, 2021-10-07:
> The easiest way to basicly play anything you want to, is by using audio loopdevices. Simply load snd-aloop via modprobe (or just add it to /etc/modules), and you are ready to go. use a terminal to [...]

I tried this method, but after i started arecord and wanted to use aplay, aplay says "aplay: set_params:1349: Channels count non available".
I don't know how to solve this, can yoou help me?