ChristopheJacquet/PiFmRds #45: Could not get audio to input

12458, 2015-11-08:
Error Dump:
Using mbox device /dev/vcio.
Allocating physical memory: size = 3403776     mem_ref = 3     bus_addr = fe625000     virt_addr = 0x7690f000
ppm corr is 0.0000, divider is 1096.4912 (1096 + 2012*2^-12) [nominal 1096.4912].
Error: could not open stdin for audio input.
Terminating: cleanly deactivated the DMA engine and killed the carrier.

Cerberus98, 2015-12-13:
Try a WAV file, or merging https://github.com/ChristopheJacquet/PiFmRds/pull/23 into your clone and recompiling, then using something like LAME to decode the file on the fly. The current code fails because it attempts to seek on a pipe (to grab the header and other meta about the audio), which is an illegal operation. The error raised is a generic catchall

12458, 2015-12-14:
Ok i will apply the patch. Thanks

12458, 2015-12-14:
How do I merge #23

Cerberus98, 2015-12-15:
The normal process is this:

// if you haven't cloned yet:
git clone https://github.com/ChristopheJacquet/PiFmRds.git

git remote add <some name you can type easily> https://github.com/Hatagashira/PiFmRds.git
git fetch --all
git merge <that same name>/patch-1

However, the branch says it has conflicts. All you really care about from that patch are the "raw" mode changes he made. 

I made that change in my repo a month back if you'd like to grab that instead: https://github.com/cerberus98/PiFmRds

Be warned I've been messing with the modulation math and I haven't really tested it out yet.

12458, 2015-12-17:
What does conflicts mean

Cerberus98, 2015-12-23:
It means that there are changes on lines in the patch that were also changed in the repository and git isn't sure which changes to keep, so it's asking the patch implementer to decide.

closed 2016-01-05 by 12458

LiuChiennan, 2018-05-22:
hello, when I use the pipe, I met the same problem, how to solve it ?please!