ChristopheJacquet/PiFmRds #47: Audio file looping endlessly

tnakk, 2016-01-13:
I am trying to play a directory full of audio files, and I wrote a bash script to play them all and shuffle between them. However, when the audio file ends, it loops to the beginning of the file instead of playing the next song. My script works flawlessly with another audio program. Please advise on how to solve this issue.

ChristopheJacquet, 2016-01-17:
PiFmRds is an experimental tool. As such, it repeats audio files, to make experiments easy even if you just have one short audio file. The program is not meant to be an "audio program". You can change that in the code if you wish.

Be advised that using the program to send out radio waves without a specific license and proper filtering is probably illegal in most countries. So please don't try to use it as a media player. And be sure not to emit radio waves when doing experiments.

closed 2016-01-17 by ChristopheJacquet

benoliver999, 2016-03-26:
One way to achieve this is to use sox, so something like

`sox -t wav /path/to/files/*.wav -t wav - | pi_fm_rds -audio -`