ChristopheJacquet/PiFmRds #48: Possible to send RDS signals without Audio?

danielmoore123, 2016-01-17:
Would it be possible to send RDS signals out and if no audio is inserted not to send any? Basically, use this alongside another transmitter, one for audio and this for RDS? Currently when you launch this without specifying audio it sends out blank audio on the frequency I want to send no audio over the frequency just RDS text, RT, PS, TA etc.

ChristopheJacquet, 2016-01-17:
It would be straightforward to modify the program to send no audio. However, if I understand correctly, you would like to use two transmitters on the same frequency. I don't think it would work : they would jam each other, but the baseband signals wouldn't add.

ChristopheJacquet, 2016-01-17:
If you want to add RDS to a non-RDS transmitter, if this transmitter is able to transmit baseband at least up to 60 kHz, then you could still reuse PiFmRds to generate the multiplex signal, play this signal on a sound card (with a sample rate > 120 kHz) and feed the results to your FM transmitter.

danielmoore123, 2016-01-17:
I thought the audio and RDS were at different ends of the FM frequency itself? Meaning one with RDS only and one with audio only should be able to work simultaneously? My non-RDS transmitter cannot have anything else plug into it you see.... and the PiFmRDS program keeps terminating after 3 hours....

closed 2016-01-17 by danielmoore123