ChristopheJacquet/PiFmRds #49: Pipe audio in via Aux lead

danielmoore123, 2016-01-18:
Would it be possible to pipe audio in via Aux lead? I feel the reason the program quits after several hours may be due to a network drop out.

jackwilsdon, 2016-01-30:
I am using [this method](http://rbpradiofm.blogspot.co.uk/2013/08/transmit-voice-from-microphone.html) which seems to work fine with PiFmRds.

danielmoore123, 2016-02-02:
That's great but it outputs in mono, I'm using a stereo USB sound card, any way to get this in stereo?

danielmoore123, 2016-02-02:
This works great! Thanks :+1: 
To clarify to the common reader, my code is:
﻿﻿
`sudo arecord -fS16_LE -r 44100 -Dplughw:1,0 -c 2 - | sudo ./pi_fm_rds -freq 100.0 -ps 'Station1' -ctl rds_ctl -rt 'You are listening to Station1' -audio -`
Using a USB Sound card: http://www.gearbest.com/development-boards/pp_176927.html?currency=GBP&gclid=CK6jqqqi2soCFRITGwodrWUN8Q

danielmoore123, 2016-02-03:
Run this overnight and get this now:

overrun!!! (at least 7.223 ms long)
Any fix for that? :-o

danielmoore123, 2016-02-04:
This is a little clearer, I installed again onto a fresh image, I get the feeling the program is designed to only run for a fixed length of time...

`overrun!!! (at least 0.503 ms long)
overrun!!! (at least 0.400 ms long)
overrun!!! (at least 0.614 ms long)
overrun!!! (at least 4.543 ms long)
overrun!!! (at least 1.988 ms long)
overrun!!! (at least 0.070 ms long)
Could not rewind in audio file, terminating
Terminating: cleanly deactivated the DMA engine and killed the carrier.
`

ChristopheJacquet, 2016-02-05:
The program is not designed that way, it's definitely a bug :) Unfortunately I have no idea what is causing it.

Thanks for the command to pipe audio, I'll include it in the README and close this bug. We can track the issue of program stopping on bug #37.

By the way you mention the "overrun" and "terminating" messages. Do they happen all at the same time?

closed 2016-02-07 by ChristopheJacquet

ChristopheJacquet, 2016-02-07:
Note: I had a look at the "overrun" message: it's a message printed by arecord when the Alsa input buffer is full. Not sure of the interaction with PiFmRds.

jackwilsdon, 2016-02-07:
Could there be any reason PiFmRds is "lagging behind" and not consuming the data at the speed ALSA is providing the data?

danielmoore123, 2016-02-27:
So..... we're killing the memory? Any way to get the program to automatically pick itself up again when it does stop?

MrSchumck, 2018-01-03:
Okay, so it seems the best way to compensate for this is to make the program accept data a bit faster. Try adding -ppm 125000 to the PiFmRds command and see if it works. Did the trick for me. If it doesn't, do some troubleshooting: if it skips backward decrease the ppm number. If it skips forward and throws an overrun, add to the ppm number. do this until it works fine. My number is 125000 with no issues for a few hours now.