ChristopheJacquet/PiFmRds #50: Static and Very Low Sound Quality on Raspberry pi 2

junhawng, 2016-02-04:
I've tried messing with the ppm, and that took care of the low pitch audio. However there is a static in all of the signals, I have tried various sound files on different frequencies. Also I think the audio output it always mono, and the output also sounds like I'm listening to music on my walki talki (very muffled). Is this a Raspberry pi 2 specific thing? I remember reading a few months back that there was super clear FM (almost station quality) for the raspberry pi do none of these work for the 2?

ChristopheJacquet, 2016-02-05:
I did not notice a difference with the Raspberry Pi 2. Having to tinker with ppm to fix audio defects doesn't sound right either. Is RDS working?

junhawng, 2016-02-06:
Hi, thank you for the reply, would there be a specific way to test the RDS? The reason I changed the ppm was because it was playing very slowly which also naturally brought down the pitch of the song. So I increased it to about 75000.

itchytweed, 2016-02-15:
Poor audio quality - low audio pitch on RPi2. Base Raspbian clock speed is 700 MHz. Was not getting a stereo beacon or RDS data through. I fixed my version of this issue by going into raspi-config and raising the clock speed to 900 MHz. Getting Stereo beacon and RDS data now as well. BTW, I am running no ppm correction.

Another issue that I have not been able to resolve by snooping in the code referencing poor audio quality is I do not see any transmission pre-emphasis of the audio, listed either at 75 us or 50 us. A FM receiver has a deemphasis circuit in it so a flat signal will sound muddy and a lack of highs to it. I fixed my version of this by using SOX and treble boost before piping it to PiFmRds.

sbruder, 2016-02-16:
On my Raspberry Pi 2 I have bad sound quality, but I also have it on my Raspberry Pi B. The Quality on the Raspberry Pi B is better than on the 2, but not good.

itchytweed, 2016-02-17:
Here is the command line that I have been using and getting some high quality FM out of this:

sox --norm [file].wav -c 2 -t wav - firfit ./75usPreEmphasis.ff | sudo ./pi_fm_rds -freq 88.5 -audio -

I am using the 'firfit' option within sox to add the necessary preemphasis for the United States so the audio does not sound so muddy using an FM receiver. The file '75usPreEmphasis.ff contains the following:

20  0.000
100 0.010
200 0.038
300 0.085
400 0.150
500 0.233
600 0.331
700 0.445
800 0.572
900 0.712
1000 0.864
2000 2.742
3000 4.742
4000 6.551
5000 8.129
6000 9.504
7000 10.711
8000 11.784
9000 12.746
10000 13.617
11000 14.412
12000 15.143
13000 15.818
14000 16.446
15000 17.033
16000 17.583
17000 18.101
18000 18.590
19000 19.054
20000 19.494
21000 19.913
22000 20.313
23000 20.696
24000 21.063

First column is the frequency in Hz and the second column is dB boost to be applied. I extended this out to 24,000 Hz as that is half of the 48000 Hz sampling frequency and will also cover 44,100 Hz sampling as well.

PepinCZ, 2016-03-03:
@itchytweed Thanks, it sounds little bit better to me.

ChristopheJacquet, 2016-03-07:
@itchytweed You're right, there's no pre-emphasis. It's tracked by bug #52 now.

A note on "sound quality": don't forget that the program is modulating a _square wave_ as a carrier. Transmitting FM using the RaspberryPi clock in the first place was a feat (cheers to the icrobotics folks!), PiFmRds adds RDS which was kind of a feat too, but don't expect too much on the audio side :) If you want good audio quality, use a proper FM transmitter, not a hack...

PepinCZ, 2016-03-07:
@ChristopheJacquet After using the sox 'firfit' option together with the file which posted @itchytweed the sound quality is actualy pretty good. I belive It can be used as home broadcasting to other room like things.

allegfede, 2016-04-29:
i suppose there is an error in the sox/pifm command.
I tested those two with success:
sox --norm -t mp3 /home/pi/Music/presentazione_Orvieto_lug_RDS_hack.mp3 -c 2 -t wav - firfit /home/pi/src/PiFmRds/75usPreEmphasis.ff | sudo /home/pi/src/PiFmRds/src/pi_fm_rds -pi 1234 -rt 'ciao agli studenti orvietani' -ps ORV-LUG -freq 88.5 -audio -

and with live audio from cheap usb sound card:
sox -t alsa hw:1,0 -c 2 -t wav - firfit /home/pi/src/PiFmRds/75usPreEmphasis.ff | sudo /home/pi/src/PiFmRds/src/pi_fm_rds -pi 1234 -rt 'ciao agli studenti orvietani' -ps ORV-LUG -freq 88.5 -audio -

Happy hacking :D

kwilium, 2016-10-08:
Hi there,

I got my best option with the following command:
sox -t mp3 http://STREAMURL.m3u -t wav -c 2 - firfit ./75usPreEmphasis.ff | sudo ./PiFmRds/src/pi_fm_rds -freq FREQ-NUMBER -audio -

Thanks for your help and enjoy

Miegl, 2017-01-11:
One thing that could improve the quality of sound but no one mentions it - the FM deviation. The default value of 25 is ok, but sounds pretty bad & static to me. However after switching to the half, 12.5, the sound improved a lot. Sure, its more quiet, but as I said, the sound quality is just better.

The deviation is defined [here].(https://github.com/ChristopheJacquet/PiFmRds/blob/master/src/pi_fm_rds.c#L192)

Simply change it to something like this:
`#define DEVIATION        12.5`

pilovis, 2017-01-17:
This is my "75usPreEmphasis.ff" file:

20 0.000
100 0.010
200 0.038
300 0.185
400 0.450
500 0.533
600 0.731
700 1.445
800 2.072
900 2.712
1000 2.864
2000 3.042
3000 4.742
4000 6.551
5000 8.129
6000 9.504
7000 10.711
8000 11.784
9000 12.746
10000 13.617
11000 12.412
12000 11.143
13000 10.000
14000 10.446
15000 10.033
16000 10.583
17000 10.101
18000 9.590
19000 8.054
20000 7.494
21000 6.913
22000 5.313
23000 4.696
24000 3.063

I reduced gain at high frequencies because of too much noise.

Also this is my command lite to rebroadcast an internet radio:

sox -t mp3 http://205.164.62.15:9010/ -t wav -c 2 - firfit ./75usPreEmphasis.ff | sudo ./PiFmRds/src/pi_fm_rds -rt 'Rochenroll Ciulers' -ps RADIO-88 -freq 88.0 -audio -

Use P2i overclock to get better performances (1000 Mhz).

pilovis, 2017-01-17:
If you want to use sox to play mp3 files or streaming internet radio (mp3), you need to install the following packages:

sudo apt-get install sox
sudo apt-get install lame
sudo apt-get install libsox-fmt-mp3

itchytweed, 2017-01-18:
I find this interesting. The way FM broadcast works is the transmitted
signal is pre-emphasized, broadcasted, de-emphasized, and out the
speakers. This was done so any received RF noise is dropped by the
de-emphasis amount for that particular frequency. Cutting the top end
transmit response is like turning down the treble on a flat conversion.

Another consideration here is that this is 75 uSec, which is the U.S.
standard for pre/de-emphasis on the FM broadcast band. I believe that
Europe's standard is 50 uSec.

Kind Regards,

Richard

pilovis, 2017-01-18:
you're right, but I had to cut the treble because of too much noise at high frequencies.

pilovis, 2017-01-19:
Anyway, I think your gain values are eccessive, factor 21 amplification at 20 KHz sounds to me quite strange.

I would suggest the following more realistic values:

<curve name="Redbook Emphasis">
<point f="20.000000000000" d="0.000000000000"/>
<point f="50.000000000000" d="0.001000000000"/>
<point f="100.000000000000" d="0.004000000000"/>
<point f="500.000000000000" d="0.096000000000"/>
<point f="600.000000000000" d="0.138000000000"/>
<point f="700.000000000000" d="0.186000000000"/>
<point f="800.000000000000" d="0.241000000000"/>
<point f="900.000000000000" d="0.303000000000"/>
<point f="1000.000000000000" d="0.370000000000"/>
<point f="2000.000000000000" d="1.290000000000"/>
<point f="3000.000000000000" d="2.430000000000"/>
<point f="4000.000000000000" d="3.540000000000"/>
<point f="5000.000000000000" d="4.530000000000"/>
<point f="6000.000000000000" d="5.380000000000"/>
<point f="7000.000000000000" d="6.100000000000"/>
<point f="8000.000000000000" d="6.690000000000"/>
<point f="9000.000000000000" d="7.190000000000"/>
<point f="10000.000000000000" d="7.610000000000"/>
<point f="11000.000000000000" d="7.950000000000"/>
<point f="12000.000000000000" d="8.250000000000"/>
<point f="13000.000000000000" d="8.490000000000"/>
<point f="14000.000000000000" d="8.710000000000"/>
<point f="15000.000000000000" d="8.890000000000"/>
<point f="16000.000000000000" d="9.050000000000"/>
<point f="17000.000000000000" d="9.180000000000"/>
<point f="18000.000000000000" d="9.300000000000"/>
<point f="19000.000000000000" d="9.400000000000"/>
<point f="20000.000000000000" d="9.490000000000"/>
</curve>

pilovis, 2017-01-19:
<curve name="Redbook De-Emphasis">
<point f="20.000000000000" d="0.000000000000"/>
<point f="50.000000000000" d="-0.001000000000"/>
<point f="100.000000000000" d="-0.004000000000"/>
<point f="500.000000000000" d="-0.096000000000"/>
<point f="600.000000000000" d="-0.138000000000"/>
<point f="700.000000000000" d="-0.186000000000"/>
<point f="800.000000000000" d="-0.241000000000"/>
<point f="900.000000000000" d="-0.303000000000"/>
<point f="1000.000000000000" d="-0.370000000000"/>
<point f="2000.000000000000" d="-1.290000000000"/>
<point f="3000.000000000000" d="-2.430000000000"/>
<point f="4000.000000000000" d="-3.540000000000"/>
<point f="5000.000000000000" d="-4.530000000000"/>
<point f="6000.000000000000" d="-5.380000000000"/>
<point f="7000.000000000000" d="-6.100000000000"/>
<point f="8000.000000000000" d="-6.690000000000"/>
<point f="9000.000000000000" d="-7.190000000000"/>
<point f="10000.000000000000" d="-7.610000000000"/>
<point f="11000.000000000000" d="-7.950000000000"/>
<point f="12000.000000000000" d="-8.250000000000"/>
<point f="13000.000000000000" d="-8.490000000000"/>
<point f="14000.000000000000" d="-8.710000000000"/>
<point f="15000.000000000000" d="-8.890000000000"/>
<point f="16000.000000000000" d="-9.050000000000"/>
<point f="17000.000000000000" d="-9.180000000000"/>
<point f="18000.000000000000" d="-9.300000000000"/>
<point f="19000.000000000000" d="-9.400000000000"/>
<point f="20000.000000000000" d="-9.490000000000"/>
</curve>

Note remove "-" to get the Emphasis values

pilovis, 2017-01-19:
curve name="Redbook De-Emphasis"
point f="20.000000000000" d="0.000000000000"
point f="50.000000000000" d="-0.001000000000"
point f="100.000000000000" d="-0.004000000000"
point f="500.000000000000" d="-0.096000000000"
point f="600.000000000000" d="-0.138000000000"
point f="700.000000000000" d="-0.186000000000"
point f="800.000000000000" d="-0.241000000000"
point f="900.000000000000" d="-0.303000000000"
point f="1000.000000000000" d="-0.370000000000"
point f="2000.000000000000" d="-1.290000000000"
point f="3000.000000000000" d="-2.430000000000"
point f="4000.000000000000" d="-3.540000000000"
point f="5000.000000000000" d="-4.530000000000"
point f="6000.000000000000" d="-5.380000000000"
point f="7000.000000000000" d="-6.100000000000"
point f="8000.000000000000" d="-6.690000000000"
point f="9000.000000000000" d="-7.190000000000"
point f="10000.000000000000" d="-7.610000000000"
point f="11000.000000000000" d="-7.950000000000"
point f="12000.000000000000" d="-8.250000000000"
point f="13000.000000000000" d="-8.490000000000"
point f="14000.000000000000" d="-8.710000000000"
point f="15000.000000000000" d="-8.890000000000"
point f="16000.000000000000" d="-9.050000000000"
point f="17000.000000000000" d="-9.180000000000"
point f="18000.000000000000" d="-9.300000000000"
point f="19000.000000000000" d="-9.400000000000"
point f="20000.000000000000" d="-9.490000000000"

pilovis, 2017-01-19:
so:

20 0.000
50 0.001
100 0.004
500 0.096
600 0.138
700 0.186
800 0.241
900 0.303
1000 0.370
2000 1.290
3000 2.430
4000 3.540
5000 4.530
6000 5.380
7000 6.100
8000 6.690
9000 7.190
10000 7.610
11000 7.950
12000 8.250
13000 8.490
14000 8.710
15000 8.890
16000 9.050
17000 9.180
18000 9.300
19000 9.400
20000 9.490
21000 10.000
22000 10.313
23000 10.696
24000 11.063

pilovis, 2017-01-19:
Miegl,

reducing the deviation form 25 to 12.5 means that you are cutting to the half the audio bandwidth.

pilovis, 2017-01-19:
If you wanto to calclulate the correct values for 50 Eu pre-emphas, use the following chart:

http://www.radiomuseum.org/forumdata/users/133/Pre_Empf_Bode_590.png

Miegl, 2017-01-20:
@pilovis I know. 12.5 just sounds little bit better with current PiFmRds.

Miegl, 2017-01-20:
I suggest trying this pull request - https://github.com/ChristopheJacquet/PiFmRds/pull/17
But keep in mind that its way better to use deviation like 75 (I wouldn't go bellow 50) with this pull request.

pilovis, 2017-01-20:
precise amplitude responses for 50uS (European) pre-emphasis

Frequence	dB
20	0.000
50	0.001
100	0.004
400	0.068
1000	0.409
2000	1.445
3000	2.761
4000	4.115
5000	5.400
6000	6.583
7000	7.661
8000	8.643
9000	9.540
10000	10.362
11000	11.120
12000	11.882
13000	12.475
14000	13.084
15000	13.656

pilovis, 2017-01-20:
**Sox firfit** is experimental, I read that the maximum knot shoud not go over 15 Khz, because it is not accurate above this frequence and may cause problems.