ChristopheJacquet/PiFmRds #52: preamphasis is missing

Blenderpics, 2016-03-04:
In order to get a decent audio quality, you really have to amplify the high frequencies.

currently i am decoding all my files, then applying an equaliser and then rendering those files back to mp3, before i use them with PiFM RDS. A rather complicated way.

I also tried using alsaequal, but because i am using loopdevices i can't get it to work, no matter what i do. (PiFm listens on Loopback,1,0; standard audio output is Loopback,0,0)

A preamphasis really should be implemented right into PiFm

are there any plans on implementing something like that? Is PiFm even still being developed?

ChristopheJacquet, 2016-03-07:
I agree, pre-emphasis is missing, it was tracked as part of bug #4. I don't have the formulas in mind, I will need to do a bit of research.

PiFmRds is an _experimental_ program whose main feature is to transmit RDS. Honestly, improving sound quality is not what keeps me going because nobody should use the program to transmit sound apart from test samples (see the README).

However I will happily apply a patch if you implement that feature :)

ChristopheJacquet, 2016-03-07:
Note: a first-order digital pre-emphasis filter is given by

y(n) = x(n) - a x(n-1)

Blenderpics, 2016-03-08:
I will give it a shot this weekend.

Blenderpics, 2016-03-12:
so i just implemented a preemphasis on a [branch](https://github.com/Blenderpics/PiFmRds/tree/preemphasis) of my fork. I just used a generall highpass-filter to implement it, even though the formular does not match the one you provided.

I have to say i have absolutely no knowledge about digital waveform filtering whatsoever, so i'm quite sure the filter is probably wrong.

Having said that i am noticing an improvement in soundquallity. Or maybe i am going nuts, who knows.

Maybe someone with more audiofiltering knowledge than me could take a look at it?

Crypter, 2021-08-04:
VERY long due, but here it is, solved:

https://github.com/Crypter/rpitx/commit/d86308ec0d8303c1bf4fbc8be2355fe052fb8dee