ChristopheJacquet/PiFmRds #53: Have you tried with Pi3 ?

pcotret, 2016-03-12:
The questio/issue is in the title :)

Blenderpics, 2016-03-12:
well, without modification, it will not run
`#if (RASPI)==1
(...)
#elif (RASPI)==2
(...)
#else
#error Unknown Raspberry Pi version (variable RASPI)
#endif`
and even if you would copy the raspberry pi 2 code part, there is very little chance that all those memory adresses are unchanged in pi 3

i am currently not owning a pi 3, so i can't say for sure, but i guess it's not working

pcotret, 2016-03-12:
Le 12/03/2016 18:29, Blenderpics a écrit :
>
> and even if you would copy the raspberry pi 2 code part, there is very 
> little chance that all those memory adresses are unchanged in pi 3
>
Yes, I know for the if/elif first part :-)
The "strange" part: there is no datasheet for BCM2837 yet (the CPU chip 
on Pi3) while this document exists for Pi1/Pi2 (in this datasheet, I'll 
find these memory addresses).

ChristopheJacquet, 2016-03-12:
No, I don't own a Pi 3 yet. With some luck there's no change compared to the Pi 2 and we'll just have to update the Makefile.

Could anyone owning a Pi 3 try this?

nickpiggott, 2016-04-09:
I have a Pi3, and there's good and bad news.

The good news is that core functionality works - I've a usable (slightly noisy) mono audio signal on the correct frequency.

The bad news is that the stereo MPX signal and RDS are missing. (The missing stereo could be the receiver switching back to mono because it's unhappy with the signal quality

I've not been able to measure the PPM, but I've tried -ppm values in increments from 100 to 2000 and none have fixed the problem.

I'll try and record the output tomorrow and get an accurate PPM figure to try with.

asp95, 2016-05-15:
Hello, i'm testing this project with the Pi3, but i have 2 problems:

1. I need to put -ppm 125396 for getting back the stereo and RDS, if i don't set this calibration error, the audio sounds only in mono and in a pitch down tone

2. I have several errors in the system while the tx runs (ex. the system audio breaks from the first tx, and i can't restore it without a reboot, and, if i move a window while the tx runs, the system has freeze).

i dont have these problems with the "markondej/fm_transmitter" project (but the audio quality is worst)

Thanks, and sorry for my english.

RegexPhilbin, 2016-06-07:
@asp95 how did you get the figure 125396?  I have a Pi Zero and this also worked for me.  Without -ppm, or a low -ppm number, everything I tried to play was slightly slow.

asp95, 2016-06-07:
@RegexPhilbin With the calibration method described in de readme, using Audacity with a record from the radio app of my Motorola phone.

RegexPhilbin, 2016-06-07:
Thanks @asp95 , I was just hoping there was a quicker and easier way.  It's interesting your value worked so well for me though.  I wonder why that would be, on a Pi Zero, which as far as I know should be pretty identical to a Pi 1, and with such a high number.  

Additional info: I'm running on the latest Retropie.  I killed all the emulation procs, and either way have plenty of headroom when PiFmRds is running.

ChristopheJacquet, 2016-11-26:
Just tested with a Pi 3. Works out of the box, just like on a Pi 2. Closing the issue, please reopen with details if needed.

closed 2016-11-26 by ChristopheJacquet

Niall7459, 2017-11-01:
@asp95 Were you able to fix the system freezing? I have the same problem since im moving my station to my pi 3 thanks!

Niall7459, 2017-11-01:
@nickpiggott Is it worth upgrading my station to my new Pi 3 then or keeping it on my Pi 2?
I want the best quality and range..

nickpiggott, 2017-11-02:
@niall7459 I've not done any measurement, but I think the Pi3 is doing a better job than the Pi2.

Niall7459, 2017-11-02:
@nickpiggott Hey could you help me on the Pi3 when I run PiFmRds its fine but then I go to move the window (or any other window) and the whole system freezes (I use VNC) but I can still connect to the webserver on the same pi, and then a minute or so after i cannot....

Please help I used to love PiFmRds

What version of Raspberry Pi are you using (Ive tested on Jessie and Stretch Pi3 but no luck with either)

Thanks

Niall7459, 2017-11-02:
I updated to stretch and it has fixed nothing, shall I update the firmware?

Niall7459, 2017-11-02:
Literally @nickpiggott Everything works, I can open folders, but as soon as I go to drag a window thats it! The system freezes and needs to be unplugged!

Niall7459, 2017-11-02:
@nickpiggott Also the green light on the Pi remains constantly lit after it freezes.

Niall7459, 2017-11-02:
It is only running the radio station at like 5% cpu, then when I move the window it all breaks down.
Just did a firmware update and it still the same...

Niall7459, 2017-11-02:
@nickpiggott Its definatley not being overloaded..

Niall7459, 2017-11-02:
@asp95 Did you ever fix the freezing on your pi?

asp95, 2017-11-02:
@Niall7459 no, i didn't use the software again

Niall7459, 2017-11-02:
Alrgiht looks like il have to use my Pi3 to host the webserver and my Pi2 for radio.

Niall7459, 2017-11-02:
@nickpiggott I am now running PiFmRds headless and when I got to sudo reboot i can still recieve the radio and the Pi never properly restarts!