ChristopheJacquet/PiFmRds #56: Output the time in an -rt message

danielmoore123, 2016-04-14:
Is there any way to have a live time output on the radiotext? (-rt)
For example:
-rt "This is Example FM, the time is --:--"

sbruder, 2016-04-15:
I think it works with
`-rt "The time is $(date +%H:%M)"`
on bash (default shell on raspbian), mksh and zsh, but if you use another shell it may not work correctly.

danielmoore123, 2016-04-15:
It works well but it doesn't change by itself.... I have a -ctl already but wish for the time to update itself without human intervention so basically, -rt "Station1 - The time is $time" and the time be changing automatically on the radiotext.... is that possible without making another -ctl?
I have an sh scrip running it, it's contents is:
`#!/bin/sh

FX='equalizer 100 50 -10 equalizer 5k 2k 0.5 compand 0.3,1 6:-70,-60,-20 -5 -65$
FREQ='107.0'
CLOCK=$(date +%H:%M)

while [ 1 ]
do
  sox -V3 -t mp3 http://playout-pc:8000/feed1 -t wav -  $FX | ./pi_fm_rds -freq $FREQ -pi FFFF -ps "Station1" -ctl  rds_ctl -rt "It's $CLOCK - You are listening to Station1" -audio -
done
`

sbruder, 2016-04-16:
It doesn't work in one script, you must run in another terminal this script:
`#!/bin/sh`
`while sleep 60;do`
`cat > rds_ctl << EOF`
`RT $(date +%H:%M)`
`EOF`
`done`
It updates your clock forever all 60 seconds.