ChristopheJacquet/PiFmRds #59: Could not allocate memory.

gi-joseph, 2016-05-11:
When I try to run the program, I get these four messages in the command line and then it just terminates:

`Using local mbox device file with major 249.
Allocating physical memory: size = 3403776     ioctl_set_msg failed:-1
Could not allocate memory.
Terminating: cleanly deactivated the DMA engine and killed the carrier.`

Obviously the issue is that it can't allocate the needed memory, but I can't figure out why.
Any help is appreciated.

Jackovic, 2016-08-08:
Seems i have the same problem here.
After runnning a long time pi_fm_rds. If i try something like :
$ sudo ./pi_fm_rds
Using mbox device /dev/vcio.
Could not allocate memory.
Allocating physical memory: size = 3403776     Terminating: cleanly deactivated the DMA engine and killed the carrier.
Any help ?
Thanx.

hisacro, 2019-11-27:
any one figured it?

hisacro, 2019-11-29:
the error is documented [here](https://github.com/JamesP6000/WsprryPi/issues/8), happens in archarm linux

wisehackermonkey, 2022-03-28:
Facing same issue

[image]
```bash
  PID TTY      STAT   TIME COMMAND
    1 pts/0    Ss+    0:04 /bin/bash /app/start.sh
 3376 pts/0    R+     0:00 ps ax
Carrier: 90.70 Mhz, VCO: 907.0 MHz, Multiplier: 47.239583, Divider: 10 
ERROR: Could not allocate memory.
Allocating physical memory: size = 4354048, Terminating: cleanly deactivated the DMA engine and killed the carrier.
```
Current theory's are:
1) a unclean exit locks the system in a state where it cant spawn that new PiFmRds instance
2) multiple copies of PiFmRds are running at the same time

It goes away when i restart the rpi3, but comes back when i restart PiFmRds  at least once