ChristopheJacquet/PiFmRds #64: Stop it from looping?

steampunko, 2016-09-09:
Hello! I have a bit of an issue. I can't get the program to stop streaming the music... help!

jesussanchez-a, 2016-09-09:
**Ctrl + C** in the command line should stop the audio from playing!

steampunko, 2016-09-09:
I also have an issue where the music plays very slowly. It wasn't doing this last night and I reinstalled PiFmRds and it still plays slowly! Even when I speed up the music via a music editor it somehow manages to make it the same exact speed. (Slow!) :(

jesussanchez-a, 2016-09-12:
That happened to me too! But I couldn't figure out what was going on. I assume its an issue with some installations. Fully remove all of the PiFmRds files and reinstall if that fails, just reinstall raspbian

steampunko, 2016-09-19:
I reinstalled and that fixed it

steampunko, 2016-09-26:
Nevermind. It's not fixed. It started acting up again and now even
reinstalling the OS doesn't work.

TechDark, 2017-05-13:
change the boot settings to console autologin