ChristopheJacquet/PiFmRds #65: "Now Playing" Data

danielmoore123, 2016-09-14:
When piping in an Icecast stream, is there any way to output the stream's attached Now Playing info, for example, attached to the stream is a line of text like: "Arctic Monkeys - I Bet You Look Good On The Dancefloor" so I would ideally like to run in this program:
`-rt "Now Playing: [icecast_text]"`
Is this possible? - Thanks in advance

krynomore, 2017-08-07:
Hey i found a way but i think it only works with shoutcast i think 
`#!/bin/sh
while sleep 60 ;do
wget -O track.txt http://Replace with your shoutcast ip:8000/currentsong?sid=1
cat >rds_ctl << EOF
RT Playing: $(<track.txt)
EOF
done`
Then chmod +x  and nohup bash whateveryousaveditas.sh &

danielmoore123, 2018-12-27:
Hey,
I'm afraid that doesn't work for me... my SH file is used to automatically restart after the program decides to 'quit' itself, so basically, a loop.

My SH file looks like this when working:
`#!/bin/sh
FX='equalizer 10 50 -10 equalizer 5k 2k 0.5 compand 0.3,1 6:-70,-60,-20 -5 -65 0.2'
FREQ='87.5'
CLOCK=$(date +%A %d %B)
TEXT='Radio Text goes here'
while [ 1 ]
do
  sox -V3 -t mp3 http://shoutcast-url -t wav -  $FX | ./pi_fm_rds -freq $FREQ -pi FFFF -ps "PS NAME" -ctl rds_ctl -rt $TEXT -audio -
done
`

danielmoore123, 2019-11-23:
Hi,
After a lot of playing, I finally got this working with Icecast! For it to work though, you will need access to a web server on the web running PHP...

Here is how I did it:

1) Under the Pi username, I set an SH script to start transmitting with audio, contents:
`#!/bin/sh

FX='equalizer 100 50 -10 equalizer 5k 2k 0.5 compand 0.3,1 6:-70,-60,-20 -5 -65 0.2'
FREQ=107.9

while [ 1 ]
do
  sox -V3 -t ogg http://airtime.jda.local:8000/airtime_128 -t wav -  $FX | ./pi_fm_rds -freq $FREQ -pi FFFF -ps "HELLO" -rt "TEST" -audio -
done`
(Source https://github.com/ChristopheJacquet/PiFmRds/issues/16#issuecomment-160032185)

2) I created a new admin user called "rds" but you can use any new admin account...

3) Under the new user, I've created another startup script in the /PiFmRds/src folder (where my ctl file "rds_ctl" is. For some reason, this additional startup script only seemed to work after it was set under a new user, I'm not sure if this is something specific to my Pi or the program... 

4) So my RDS SH file contains as follows:
`#!/bin/bash
# Now Playing RDS
cd /FM/PiFmRds/src/
while sleep 10 ;do
RT=$(curl https://www.url-to-php-file-not-icecast.com/file.php)
cat >rds_ctl << EOF
RT $RT
EOF
done
` 
Credit: Solution based on: https://github.com/ChristopheJacquet/PiFmRds/issues/65#issuecomment-320808330 - Follow that if you are on SHOUTcast.

5) I make that file start on boot as the rds user and not pi, that works but can take a few seconds longer on boot time.

6) Next, create a new PHP file on a web server, note the PHP server needs to have cURL enabled.

7) Contents of the PHP file: 
`Now Playing: <?php
$ch=curl_init();
curl_setopt($ch,CURLOPT_URL,'https://icecast-server:8000/status-json.xsl?mount=/stream');
curl_setopt($ch,CURLOPT_COOKIE, session_name() . '=' . session_id()); 
curl_setopt($ch,CURLOPT_CONNECTTIMEOUT,5);
curl_setopt($ch,CURLOPT_RETURNTRANSFER,1);
curl_setopt($ch,CURLOPT_FOLLOWLOCATION, false ); 
    $data = curl_exec($ch);
    preg_match('#,"title":"(.+?)","dummy"#is', $data, $matches);
    foreach ($matches as &$match) {
        $match = $match;
    }
        echo $matches[1];
curl_close($ch);
?>`

8) Quickly go to https://your-url/phpfile.php in a browser to see the Icecast Now Playing data show

9) Reboot the Pi, assuming both the Pi & rds users have their startup scripts to start the carrier and web stream and then wait for the play.sh file to start making ctl >rds_ctl commands!

Not the best solution, but quite a smart workaround!

closed 2019-11-23 by danielmoore123