ChristopheJacquet/PiFmRds #68: Play folder ?

blocksberg, 2016-10-04:
Hey great works, and it works very well...

..atm i make big *wav Files (1.wav + 2.wav +3.wav ......build some files to one wav file) with audacity to play some songs in a loop. I search for a code to play all files in one folder....
Sure, audacity method works, but its a lot of work to do this....A simple code like ./pifmrds -audio "/folder/././" to play all files in this one folder can do this work better.

br
Blocksberg on Potatojam

mitchellurgero, 2016-12-09:
using bash scripting, you can use a for loop to list all files in a folder, and then play them using pifmrds.

``` bash
#!/bin/bash
for file in /path/to/wav/files/{.,}*; do  ./pi_fm_rds -audio "$file"; done

```
Or something to that affect.

Niall7459, 2017-02-07:
Use VLC Media play, tis the best!

ChristopheJacquet, 2019-08-01:
I hope the suggestions above helped you achieve what you wanted. I'm closing this issue as Pi-FM-RDS is not a media player, so playing a folder would be beyond its scope.

closed 2019-08-01 by ChristopheJacquet