ChristopheJacquet/PiFmRds #72: If you install rpitx AM-FM-SSB-VCO modulator, PiFmRds won't work anymore!

pilovis, 2017-01-18:
After installing rpitx, my PiFmRds stopped working, or better it started transmitting on a wrong frequence and on the wrong GPIO pin. :(

This is because rpitx creates "/boot/dt-blob.bin" file, that changes the default pin configurations of the Raspberry (persistent modification after reboot)!
At least the programmer should have warned the user before starting the installation of this program.
Also, weak programmers do not write instructions how to uninstall their programs.

Anyway, to remove Rpitx and to get PiFmRds working again, follow these steps:

rm /boot/dt-blob.bin
rm /usr/bin/pifm
rm /usr/bin/piam
rm /usr/bin/pisstv
rm /usr/bin/pissb
rm /usr/bin/pifsq
rm /usr/bin/rpitx

pilovis, 2017-01-18:
reference: https://www.raspberrypi.org/documentation/configuration/pin-configuration.md

TheMakaf, 2017-10-14:
Thanks a lot! After a 3 hours of tests, i finally find answer why i have bugs in radio transmission.

xMateMCPE, 2018-07-21:
Thanks! I had the same problem, was searching for about 2 hours a fix.