ChristopheJacquet/PiFmRds #73: use mpd/mpc with PiFmRds to rebroadcast internet radios and/or MP3 playlist

pilovis, 2017-01-19:
This is the audio section for /etc/mpd.conf:

audio_output {
type "pipe"
name "PiFm"
command "sox -t raw -esi -b32 -c2 -r44100 - -t wav - firfit /root/75usPreEmphasis.ff | sudo /root/PiFmRds/src/pi_fm_rds -freq 88.0 -rt 'Rochenroll Ciulers' -ps RADIO-88 -audio $
format "44100:32:2"
}

note: eventually adjust the paths on the above.

pilovis, 2017-01-19:
sudo apt-get install sox
sudo apt-get install lame
sudo apt-get install libsox-fmt-mp3

pilovis, 2017-01-19:
this is my 75usPreEmphasis.ff file , not a real 75uS emphasis, but it gives the best audio output:

_20 0.000
50 0.001
100 0.004
500 0.096
600 0.138
700 0.186
800 0.241
900 0.303
1000 0.370
2000 1.290
3000 2.430
4000 3.540
5000 4.530
6000 5.380
7000 6.100
8000 6.690
9000 7.190
10000 7.610
11000 7.950
12000 8.250
13000 8.490
14000 8.710
15000 8.890
16000 9.050
17000 9.180
18000 9.300
19000 9.400
20000 9.490
21000 10.000
22000 10.313
23000 10.696
24000 11.063_

pilovis, 2017-01-19:
The best would be to get the current playing song name/artist from MPD with "mpc current" command and inject it to RDS "rt"

pilovis, 2017-01-19:
I have an idea, what if we create a network of PiFmRDS transmitter all connected to a single radio net stream?
Each transmitter uses very low transmitting power but connecting many togheter we could create a hudge coverage, like the WiFi fre hot spots

pilovis, 2017-01-19:
By connecting an antenna like this one, the coverage range will increase up to 500 mt on open air:

https://www.amazon.com/Stellar-Labs-30-2435-Outdoor-Omnidirectional/dp/B00DHHOZBI

connect the hot pole to Pin 4 of the Raspebrry and the cold pole to GND, also connect GND of Raspnberry to ground (heart ground).
A filter from raspberry to the antenna, to cut armonics frequences is absolutely needed.

pilovis, 2017-01-20:
If you want to make the transmitter more robust and less sensitive to micro network outages, add these lines at the end of the /etc/mpd.conf file:

samplerate_converter "internal"
audio_buffer_size "8192"                                                                                                                             
buffer_before_play "10%" 

To get more infos on MPD/MPC see my specific how-to on OpenWRT forum:
[HowTo] Openwrt cheap Digital Stereo WiFi Internet Radio & MP3 Player
https://forum.openwrt.org/viewtopic.php?id=49013

Santanix, 2017-02-13:
Hello!
I'm new to Pi but I manage to trasmit mpd music to FM. What I cannot is with your code because I can't find the file "75usPreEmphasis.ff". I hear some background noise and in another thread you say the sound get better with PreEmphasis. My cod working is:

audio_output {
        type            "pipe"
        name            "PiFm"
        command         "sox -t raw -b 32 -c 2 -r 44100 -e signed-integer - -t wav -b 32 -c 2 -r 44100 - | sudo /home/pi/PiFmRds/src/pi_fm_rds -audio -"
        format          "44100:32:1"
}

Can you explain how I can get PreEnphasis working with your file PreEnphasis?

Thank you!

spabinger, 2017-03-13:
Add this parameter to your sox command (see first post)
`- firfit /root/75usPreEmphasis.ff`

Santanix, 2017-03-13:
Thanks for the reply! 
I manage to get it working with the code equal to yours, I only changed the path to the file 75usPreEmphasis because I couldn't find the file so I created one in dir of PiFmRds. 
command "sox -t raw -esi -b32 -c2 -r44100 - -t wav - firfit /pi/PiFmRds/75usPreEmphasis.ff | sudo /pi/PiFmRds/src/pi_fm_rds -audio -"
format "44100:32:2"
Is ok making the PreEmphasis file? ( I make a test file, changed the name and termination like above, and inside are the numbers you posted here)
Again, Thank You!

pilovis, 2018-09-10:
see here for updated how-to:

https://github.com/Miegl/PiFmAdv/issues/37