ChristopheJacquet/PiFmRds #75: Usage on Raspberry Pi 3 plays example .wav files to slow

pierreverbakel, 2017-01-29:
I have tried to compile and run the code on a Raspberry Pi 3

My radio receives the signal, but the voice is very slow like an 45/min disc running 33/min (Yes I was alive when those exist)
In my /boot/config.txt I have:
----
force_turbo=1
----

Also I do not see any RDS information. But that might be linked to the original root-cause.

What do I need to change in the code to let it run faster?

Thanks!

hyx0329, 2017-02-07:
Try to run the program in cli instead of gui. In this way you may broadcast your song at a normal speed.
I don't know why but this is the only solution works on my pi3.
else see #50

Miegl, 2017-02-20:
It might be problem with the clock. Try calibrating it a bit. Guide is in https://github.com/ChristopheJacquet/PiFmRds/blob/master/README.md

pierreverbakel, 2017-02-20:
I understand that there might be a calibration needed, but the correction I need is in the order of a factor two (2) !! on my Raspberry Pi 3
I have measured with a stopwatch the time in between the start of the horn and the next start of the horn using:
./pi_fm_rds -audio pulses.wav

There must be something else I can change in the code to speed it up the playback this factor 2?
Or should and can I use the calibration procedure for this?

Miegl, 2017-02-21:
Try calibrating it (you can even try to use some random ppmand tell if its better).
I don't have Rpi3 so can't help you much with code.

hyx0329, 2017-02-23:
@pierreverbakel
i should notice that: KILL xserver on tty1
or just set the pi to boot to console
this will 99% help you
(and i don't know why)

Janevski, 2017-04-02:
I'm experiencing the very same problem.

I added radio script execution hook into rc.local, booted pi into shell only and it seems a lot better!
If i enable the gui, everything works fine again on reboot up to the point where i type in username, password and login into xfce - the script fails and if i reinitialize it i either get audio resources busy or the same low pitch and audio cuts because of it voice problem.

I've tried invoking scripts and executables with stdbuf -o0 so the pipe buffers would flush, i thought it might be the way a gui shell piping works, but no, doesn't seem to help at all.

ALTracer, 2018-01-13:
My RPi3 requires ppm=125000 to broadcast music at perceivably same speed as source. Why 5/4?
Not overclocked, 1200/600 MHz CPU.

ChristopheJacquet, 2019-08-01:
I checked on my RPi3, speed is correct whether using force_turbo (scaling_cur_freq = 1200k) or not (scaling_cur_freq = 600k).

I'm using stock Raspbian, no GUI. What's different in your setups?

ChristopheJacquet, 2019-08-01:
I found a recommendation to add `gpu_freq=250` to `/boot/config.txt`. Source and explanations: https://github.com/SaucySoliton/PiFmRds/commit/41c6523fc65894a25efa3e1b8d921201679da7ca

Does this work for you?

miegl, 2019-08-02:
@ChristopheJacquet that is only needed when using the GPU for modulation (changing the frequency of PLLC and/or PLLA). PiFmRds works by dividing the frequency of PLLD, so not needed in this case.

ALTracer, 2019-08-02:
> I'm using stock Raspbian, no GUI. What's different in your setups?
@ChristopheJacquet ping me if you're addressing me.
I am not actively using PiFmRds at the moment, but here's my configuration. Raspbian Stretch GUI from NOOBS, upgraded since Jessie. Yes, there are `Under-voltage detected! (0x00050005)` warnings in dmesg due to an external USB HDD attached. Even an external powered USB3.0 hub doesn't quite help. I may check radio behaviour again over this weekend. Please tell exact length of GPIO antenna as I'm using a FLIRC aluminum case clone.
```
altracer@ALT-RPi3:~ $ grep -v "^#" /boot/config.txt
disable_overscan=1
hdmi_group=2
hdmi_mode=35
dtparam=sd_overclock=100
dtparam=audio=on
dtparam=watchdog=on
enable_uart=1
start_x=0
dtparam=spi=on
dtparam=i2c_arm=on
dtparam=i2c0=on
dtparam=i2c1=on
```

scratchermatch, 2020-03-04:
Super dumb workaround:
Navigate to where your file is stored
then type:
sox input.mp3 output.mp3 speed 2.0
Input is the song that is playing slowly
Output is the new file name (add like a x2 to the end or something so you don't get confused)
It'll take a few seconds for each file.
This literally makes the song twice as fast, and since it plays at 0.5% speed, this makes it normal.
Source: https://ankitshah009.blogspot.com/2016/04/sox-sound-exchange-how-to-use-sox-for.html

dsdhinesh1987, 2021-06-06:
Best Audio Output in PI 3B+

tuned Speed, volume, Channel and firfit option.

sox --norm -v 2.0 -t mp3 tamil.mp3 -t wav -c 1 - firfit ./75usPreEmphasis.ff speed 2.0 | sudo ./pi_fm_rds -rt 'Sakthi FM' -ps RADIO-102 -freq 102.0 -audio -

LumineRose, 2021-08-04:
**Solving the problem** at Rpi 3b (and 3b+ and maybe 3) = **set the ppm value to 1000000**
In my case, the command looks like this and everything plays perfectly ./pi_fm_rds -freq 101.5 -audio filename.wav -ppm 1000000

ryanginnelly252, 2022-01-31:
> **Solving the problem** at Rpi 3b (and 3b+ and maybe 3) = **set the ppm value to 1000000** In my case, the command looks like this and everything plays perfectly ./pi_fm_rds -freq 101.5 -audio filename.wav -ppm 1000000

This worked!

Mifaxav, 2023-04-17:
sox --norm -v 2.0 -t wav /home/mifa/rpitx/never-gonna-give-you-up-official-music-video.wav -t wav -c 1 - speed 1.0 | sudo ./pi_nbfm -freq 434.0 -audio -
 
 This fixed it for me