ChristopheJacquet/PiFmRds #79: Audio output stops working and crackle

Flavien06, 2017-02-10:
Hi,
I use pifm in the morning to listen radio with this command
` sudo sox -t mp3 $urlradio -t wav -r 44100 -c 2 - | sudo ./pi_fm_rds -freq $FREQ -audio - -ps $radio -rt 'LaGrosseRadio'`

And I stop with `sudo killall sox`

But when I start PiFm, audio output stops working.
If I play a music with MPD I have crackle in audio output.
To fix it I reboot the RPi.
Do you have idea of source of this bug? And how to fix it?

Thank

3c4ng1s, 2017-02-27:
I must mostly apologize for my English in advance if I commit any mistake, and for the same reason that I'm not quite sure if I'm having the same problem. 

In my case the issue is that sound may or not start but it never survive more than 8 or 10 seconds Top, after that it just stop transmitting but the process itself it's not stopped, just the output of sound.

I'm using a raspberry pi 3 with a 3,5" DFRobot tft and a 1 meter (~3,5 ft).

Any help will be great. Thanks.

Niall7459, 2017-04-01:
Try "sudo rmmod w1-gpio" when you restart your pi.

Flavien06, 2017-04-05:
@Niall7459
> Try "sudo rmmod w1-gpio" when you restart your pi.
I have this error : 
`rmmod: ERROR: Module w1_gpio is not currently loaded`

I made a new install on RPi and always the same problem.
When I want to reboot RPi (if PiFmRDS works) I have no answer froml the RPi (I unplug it!)

@3c4ng1s I did not fully understand your problem. What is your native language?
I use a loop for PiFM [https://github.com/ChristopheJacquet/PiFmRds/issues/16#issuecomment-160032185](url)

`while [ $var_PIFM = "MARCHE" ] # Boucle pour erreur: cleany deactivated the DMA engine and killer the carrier
do
	. ./MES_VARIABLES.txt
         mpg123 --buffer 2048 --no-icy-meta -q -s $var_URLRADIO | sox -t raw -b 16 -e signed -c 2 -r 44100 - -t wav - equalizer 11800 0.7q 20 | sudo /home/pi/PiFmRds/src/pi_fm_rds -rt 'LaGrosseRadio' -pi 0528 -ps $var_URLRADIOTYPE -freq $freq -audio -
do`

Niall7459, 2017-11-12:
Pi 3?

Flavien06, 2017-11-12:
No Pi 1 B