ChristopheJacquet/PiFmRds #82: Only Play Audio One Time Without Looping

convolutionbbs, 2017-03-04:
I would like to use pifmrds to only play through a .wav file one time.  I have a sequence of files that I would like to play in different orders.  I saw another issue in here where the solution was "press ctrl-c".  I want to run this in the background without me having to type this.  One thought I had was to get the PID after executing, then run a kill command.  However, I won't know when the audio is complete from the first playthrough.

I tried running: sudo ./pi_fm_rds -audio -

but I get Error: could not open stdin for audio input.  (raspberry pi 2 with latest raspbian)

Even if that worked, I guess I would run other commands one at a time to pass to the standard audio.

I saw another post regarding using a for loop in a bash script - but pifmrds keeps iterating through the first .wav file it comes across.

Thanks!

tchiobtcha, 2017-03-08:
yeah same problem with me 
i tried to change the c code and re"make" the code to get the new PiFmRds but with no sucess
hope there is a solution for that

scitronboy, 2018-03-30:
You can solve this by using the stdin option:

cat sound.wav | sudo ./pi_fm_rds -audio -