ChristopheJacquet/PiFmRds #87: More RDS (PTY)

Niall7459, 2017-03-29:
Can we have RDS PTY (Program Type)

Miegl, 2017-03-29:
Hi,
you can try my fork "PiFmAdv", which has PTY implementation.
https://github.com/Miegl/PiFmAdv

Niall7459, 2017-03-29:
NICE THATS AWESOME

preterive, 2017-12-03:
very good and with better quality.

KubaPro010, 2023-11-11:
for anyone that pifmadv isn't working you can check my fork of rpitx "rpitx2" (it has pifmrds, with the same rds encoder as pifmadv has), for install just follow the normal rpitx install

ryanginn, 2023-11-13:
> for anyone that pifmadv isn't working you can check my fork of rpitx "rpitx2" (it has pifmrds, with the same rds encoder as pifmadv has), for install just follow the normal rpitx install

Nice one, must take a look later. Does changing PS and RT while running (-ctl) work in this branch? I've had mixed success with it on PiFMAdv and PiFMRDS.

Ryan

andimik, 2023-11-13:
Is there any branch which can handle mpx as input?

KubaPro010, 2023-11-13:
> Nice one, must take a look later. Does changing PS and RT while running (-ctl) work in this branch? I've had mixed success with it on PiFMAdv and PiFMRDS.
>
> Ryan

the control pipe has more features, and is working (when its enabled with the `-ctl` arg)

KubaPro010, 2023-11-14:
> > > > for anyone that pifmadv isn't working you can check my fork of rpitx "rpitx2" (it has pifmrds, with the same rds encoder as pifmadv has), for install just follow the normal rpitx install
> > > 
> > > 
> > > Nice one, must take a look later. Does changing PS and RT while running (-ctl) work in this branch? I've had mixed success with it on PiFMAdv and PiFMRDS.
> > > Ryan
> > 
> > 
> > the control pipe has more features, and is working (when its enabled with the `-ctl` arg)
> 
> Ah fair play that's awesome! Must give it a shot.
> 
> > Is there any branch which can handle mpx as input?
> 
> This would be real handy with tests with Stereo Tool
> 
> Ryan

There isn't a branch, i have implemented one of the stereo tool's features if you wanna know (rds volume)

ryangontv, 2023-11-14:
> There isn't a branch, i have implemented one of the stereo tool's features if you wanna know (rds volume, it cant take raw mpx)

Thanks for clarifying, RDS volume is really making me interested. must give it a go in an hour or so.

Many thanks!
Ryan

KubaPro010, 2023-11-15:
implemented a stereo pilot volume now

> > There isn't a branch, i have implemented one of the stereo tool's features if you wanna know (rds volume, it cant take raw mpx)
> 
> Thanks for clarifying, RDS volume is really making me interested. must give it a go in an hour or so.
> 
> Many thanks! Ryan

andimik, 2023-11-15:
@KubaPro010 

Could you try to only upload fully working commits? I've seen your repository's history and guess you are testing your own code for the public.

KubaPro010, 2023-11-15:
it's experimental, so something may not work