ChristopheJacquet/PiFmRds #93: Makefile44 error

d3bugg3d, 2017-08-02:
I keep on getting a Makefile:44: recipe for target 'pi_fm_rds.o' failed
make: *** [pi_fm_rds.o] Error 1.

I'm SSH'ing into my Pi from Putty. Any help?

thedarkeye, 2017-08-03:
Have you got all the dependencies needed? Like  `libsndfile1-dev`

d3bugg3d, 2017-08-03:
Yeah I typed that command in on my Pi. Should I retry it? What command do I enter?

d3bugg3d, 2017-08-19:
Can anybody comment on this? Does this hack still work for other people?

thedarkeye, 2017-08-20:
I've tested it on my Pi 2 B+ with Raspbian Jessie Lite and it works fine.
I don't know why it wouldn't work for you, if you haven't already try updating with `apt-get update && apt-get upgrade`.

d3bugg3d, 2017-08-20:
When I try installing the update or whatever it says failed to install.

⁣Sent from Blue ​

d3bugg3d, 2017-08-21:
https://null-byte.wonderhowto.com/how-to/hack-radio-frequencies-hijacking-fm-radio-with-raspberry-pi-wire-0177007/

Trying to follow this guide. It won't install the libsnd accordingly. I
think I need to start from scratch again. Do you have the proper steps?

d3bugg3d, 2017-08-21:
> frequencies-hijacking-fm-radio-with-raspberry-pi-wire-0177007/

I don't know which commands to use to upgrade it or install it or whatever.
Then it says that too much of the disc memory is taken or some shit like
that.

ChristopheJacquet, 2017-08-21:
I can advise you to start from a fresh Raspbian install to do your experiments with PiFmRds. Then just install libsndfile1-dev (apt-get update && apt-get install libsndfile1-dev) and you should be all set.

Closing this issue as it's not directly related to PiFmRds.

closed 2017-08-21 by ChristopheJacquet

d3bugg3d, 2017-08-22:
Ok, whenever I type in the git clone
https://github.com/ChristopheJacquet/PiFmRds.git
cd PiFmRds/src
make clean
make

I get an error saying "Makefile:44: recipe for target 'pi_fm_rds.o' failed
make: *** [pi_fm_rds.o] Error 1