ChristopheJacquet/PiFmRds #101: Antenna to extend range?

Brandon4466, 2018-01-15:
Are there any suggestions for an antenna to use with this? Others experiences?

I have found anything like this: https://www.amazon.com/dp/B01HTDKAWS/ref=cm_sw_r_cp_apa_JsbxAbGW04SRV

Would something like this work? Does anyone know how you would connect something like this best to the GPIO?

Basically just the best way to boost the radius. Any suggestions would be great! What do you guys use for an antenna?

Thanks!

hex007, 2018-01-15:
There are various restrictions in place for FM broadcasting based on your country. Please check them before broadcasting with higher power antennas.

https://github.com/ChristopheJacquet/PiFmRds#warning-and-disclaimer

Artoria2e5, 2018-02-20:
Assuming you know what you are doing, there's always https://ham.stackexchange.com/7365 …

MinePro120, 2018-03-31:
A 30cm copper wire with a female jumper cable mounted on the one side should do the trick. That's what I use.

closed 2019-07-31 by ChristopheJacquet