ChristopheJacquet/PiFmRds #103: Dynamic Program Service Name

MinePro120, 2018-03-31:
Hi,
I was wondering whether it is possible to display multiple -ps texts. Like the ones real radio stations transmit eg :
Rock 
Radio
97,6
Stay
Tuned

Miegl, 2018-03-31:
You can use a FIFO pipe for that. 
```
mkfifo rds_ctl
sudo ./pi_fm_rds -ctl rds_ctl
```
```
cat >rds_ctl
PS Rock
PS Radio
```

MinePro120, 2018-04-01:
Thank you very much. As I am a complete noob at this, do I just have to run those commands from the PiFmRds directory?

Miegl, 2018-04-01:
Just run these commands in the directory with pi_fm_adv.

MinePro120, 2018-04-01:
I got it. Changing the ps in real time is not what I am looking for. I want it to be automatic and loop the same ps's again and again without me having to do anything. Is it possible?

Miegl, 2018-04-01:
https://bash.cyberciti.biz/guide/Main_Page
https://www.cyberciti.biz/faq/bash-for-loop/

collinator24, 2019-08-24:
I wanted to do this as well. Did anyone figure it out?

collinator24, 2019-08-24:
did you figure it out?

ChristopheJacquet, 2019-08-24:
Currently you have to send a "PS" command to the FIFO each time you want to change the PS. You can do a loop, e.g. in bash, to produce a cyclic effect.
I added the "enhancement" label: if I have time I may add this feature directly to Pi-FM-RDS.

collinator24, 2019-08-24:
That's cool. For some reason, when I made my script it would stop working
after the first one. It doesn't let me cat the file when the pifmrds is
running. If I cat the file, then run it it works but because my script does
it, sleeps, then does a different command it doesn't work. Maybe I just
have bad luck.

MinePro120, 2019-08-24:
I created the script some time ago, it worked fine.

closed 2019-08-24 by MinePro120

WisseHesNL, 2019-08-25:
Can you send the script?

MinePro120, 2019-08-25:
#!/bin/bash

while true
do
	echo 'PS Tempo FM' | cat >rds_ctl
	sleep 5
	echo 'PS Quality' | cat >rds_ctl
	sleep 5
	echo 'PS Music' | cat >rds_ctl
	sleep 5
	echo 'PS 108 MHz' | cat >rds_ct
	sleep 5
	echo 'PS Stay' | cat >rds_ctl
	sleep 5
	echo 'PS Tuned!' | cat >rds_ctl
	sleep 5
	echo 'PS '$(date +"%H:%M") | cat >rds_ctl
	sleep 5
        echo 'PS '$(date +"%d/%m/%y") | cat >rds_ctl
	sleep 5
done
exit 0

There you go. I've added some extras too.

WisseHesNL, 2019-08-25:
Hmm, apparently `echo 'PS radio' | cat >rds_ctl` only works when you restart PiFmRds everytime you change it, but 
```
pi@pi-fm:~/PiFmRds/src $ cat >rds_ctl
PS oof
```
Does work

MinePro120, 2019-08-25:
I didn't experience that one, neither I remeber the reason I used 'echo'. But if it works...

MinePro120, 2019-08-25:
<code> test ] </code>

WisseHesNL, 2019-08-25:
But doesn't work

WisseHesNL, 2019-08-25:
What would you use instead of 'echo'?

MinePro120, 2019-08-25:
Simply cat >rds_ctl foo

WisseHesNL, 2019-08-25:
```
pi@pi-fm:~/PiFmRds/src $ cat >rds_ctl PS foo
cat: PS: No such file or directory
cat: foo: No such file or directory
```

WisseHesNL, 2019-08-25:
using 'PS foo' also does not work btw

MinePro120, 2019-08-25:
Now I remeber the reason I used echo. But I didn't encounter any peoblem neither with the echo command nor with the script anyways. What RDS receiver are you using? I used my phone and noticed some latency and other issues. But that was the case with every station.

WisseHesNL, 2019-08-25:
I tried it on my Pure Evoke H4 radio and my old Sony ICF-M50RDS. But i dont even see anything in the terminal like `PS set to:`

MinePro120, 2019-08-25:
So the whole script doesn't work at all?

WisseHesNL, 2019-08-25:
I think so, but i tried other things like 
```
cat >rds_ctl <<EOF
PS TEST
EOF
```
And it does change the text in the FIFO file, but PiFmRds doesn't see it for some reason.

MinePro120, 2019-08-25:
I don't know, it worked in my case. Maybe @ChristopheJacquet can help.

WisseHesNL, 2019-08-25:
Did you use Pi_fm_adv or just PiFmRds because Miegl said you have to run it in the pi_fm_adv directory

MinePro120, 2019-08-25:
What is Pi_fm_adv? I ran it inside PiFmRfs/src.

WisseHesNL, 2019-08-25:
Oh okay, Miegl made a more advanced version of PiFmRds

miegl, 2019-08-25:
Can confirm; PiFmRds isn't able read from the pipe after a while. Haven't figured out why yet.

MinePro120, 2019-08-25:
Oh wait, I just remembered that I've been using F5OEO's version of PiFmRds. Oops.

WisseHesNL, 2019-08-26:
THANKS!! It works!!

MinePro120, 2019-08-26:
What works?

WisseHesNL, 2019-08-26:
The script does work with F5OEO's version of PiFmRds :)

MinePro120, 2019-08-26:
And you gained some sound quality improvements too. But under no circumstances should you put any antenna on that thing. I even built a low-pass passive filter, but it was in vain.

WisseHesNL, 2019-08-26:
(i'm a complete noob) Why should you not put an antenna on it?

MinePro120, 2019-08-26:
Because obviously Raspberry Pi is not made for this. The sine fm radio waves are generated by the pwm function of the pi, which creates huge harmonics, namely other waves at many different frequencies. I did some tests myself. When I set the frequency to 108 Mhz, there were signals at 200 and 300 Mhz stronger than the original! I also found the range to be ~500m with a whip antenna, simply a wire connected to the pin, without any obstacles. So it is certain that it will interfere with other frequencies, possibly aircraft communications too. Not a good idea to leave it running for a long time.