ChristopheJacquet/PiFmRds #104: Change volume

wissehes, 2018-04-02:
Hi, sorry if this isn't the right place to ask this, but is there a feature to change the volume of this command:
`sudo arecord -fS16_LE -r 44100 -Dplughw:1,0 -c 2 -  | sudo ./pi_fm_rds -audio -`
because the volume of the sound is very low...

thanks

Marrley, 2020-06-25:
Have You Fixed It? Im Having The Same Problem

TheChicken14, 2020-06-26:
No, I couldn't find a fix, but it's not a big issue if you just turn the volume on your radio up I guess

KubaPro010, 2023-11-11:
you could use my fork of rpitx (and it does have pifmrds, with better rds encoder and better control over transmission,but for now the code is made for the pi 3 but can be very easy changed to a another rpi), you can specify a `-audiogain` arg or control the gain with the fifo (send to the fifo: `GAI {gain level}`, also gain is just how many times is the audio louder)