ChristopheJacquet/PiFmRds #106: Transmitting simultaneously on different frequencies? Emergency Alarm system

JsBergbau, 2018-04-10:
Is it possible to transmit for example at 89 92 95 97 MHz simultaneously? 
I think about some kind of dithering, so fast switching between the frequencies and have wider steps in the generated signal. Of course that would massively affect quality and probably the more parallel frequencies the worse the quality gets. However it could be used as an emergency alarm system. You take the frequencies of the main radion stations in your area and broadcast on their frequency your alarm message. The quality would be bad of course, but should be sufficient for a simple voice warning. Yeah and I know that you have to be quite near the receiver because of the other signal, but still it would work and could be a very useful use case.

JonQuan2, 2018-07-08:
That would be a good idea but the raspberry PI doesn't have the bandwidth to do that. You might be able to work something out by using multiple GPIO Pin outputs. Pin 18 I think.

Also I would not use this to broadcast EAS messages on active frequencies. That's just asking for the FCC to shut you down.

btweb, 2018-07-10:
Also consider that you will probably run in to intermod trouble either trying to push all frequencies through the same active element, or several elements close together. I'm not an expert by any means but I suggest you look up intermod and how to deal with it if you do manage to do this. Usually you use filters to isolate any harmonics, of which there will be plenty with a PWM generated wave form :-)

ChristopheJacquet, 2019-07-31:
Closing, I think the original question was answered above.

closed 2019-07-31 by ChristopheJacquet