ChristopheJacquet/PiFmRds #107: overrun problem is showing

rakshit07tiwary, 2018-04-12:
after running the command pi transmitting the signals but the song is playing very slow.it is showing like overrun (atleast ____ ms long).please help me in resolving this problem.

tiny4jinxin, 2018-04-21:
i have the same problem at first , i found the spi LCD screen is using the
GPIO  at the same time, after uninstalled the driver of LCD screen, the
song playing in right speed.
you can check if there is other application is using the GPIO, hope this
help.

mtribiere, 2018-10-31:
I have the same problem here but wanna use an 7" touch screen anyway. Anyone as a solution ?

mtribiere, 2018-11-07:
Problem almost solved with this post : https://github.com/ChristopheJacquet/PiFmRds/issues/75#issuecomment-357476159

rakshit07tiwary, 2018-12-02:
> Problem almost solved with this post : #75 (comment)

Yeah i have an solution just mail me ur mail id and i will give you the
link you can copy code from that.
Thanks