ChristopheJacquet/PiFmRds #115: Rasberry Pi 3B+ install issues

basimci, 2018-12-08:
Tried setting up PiFmRDS on RP 3B+ and i am getting this warning 
anyway to fix?

root@kali:~# git clone https://github.com/ChristopheJacquet/PiFmRds.git
Cloning into 'PiFmRds'...
remote: Enumerating objects: 311, done.
remote: Total 311 (delta 0), reused 0 (delta 0), pack-reused 311
Receiving objects: 100% (311/311), 3.39 MiB | 169.00 KiB/s, done.
Resolving deltas: 100% (191/191), done.
root@kali:~# cd PiFmRds
root@kali:~/PiFmRds# ls
doc  LICENSE  README.md  src
root@kali:~/PiFmRds# cd src
root@kali:~/PiFmRds/src# ls
control_pipe.c      generate_waveforms.py  pi_fm_rds.c  sound_22050.wav
control_pipe.h      mailbox.c              pulses.wav   sound.wav
fm_mpx.c            mailbox.h              rds.c        stereo_44100.wav
fm_mpx.h            Makefile               rds.h        waveforms.c
generate_pulses.py  noise_22050.wav        rds_wav.c    waveforms.h
root@kali:~/PiFmRds/src# make clean
rm -f *.o
root@kali:~/PiFmRds/src# make clean
rm -f *.o
root@kali:~/PiFmRds/src# make
gcc -Wall -std=gnu99 -c -g -O3 rds.c
rds.c: In function ‘set_rds_rt’:
rds.c:240:5: warning: ‘strncpy’ specified bound 64 equals destination size [-Wstringop-truncation]
     strncpy(rds_params.rt, rt, 64);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rds.c: In function ‘set_rds_ps’:
rds.c:247:5: warning: ‘strncpy’ specified bound 8 equals destination size [-Wstringop-truncation]
     strncpy(rds_params.ps, ps, 8);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -Wall -std=gnu99 -c -g -O3 waveforms.c
gcc -Wall -std=gnu99 -c -g -O3 rds_wav.c
gcc -Wall -std=gnu99 -c -g -O3 fm_mpx.c
gcc -o rds_wav rds_wav.o rds.o waveforms.o fm_mpx.o -lm -lsndfile
root@kali:~/PiFmRds/src#

basimci, 2019-01-08:
Issue with Kali os
Working on Raspberry OS

closed 2019-01-08 by basimci

saintofinternet, 2019-01-15:
any idea on how to solve it??