ChristopheJacquet/PiFmRds #118: sudo ./pi_fm_rds not found

DylanHP, 2018-12-23:
help me the command sudo ./pi_fm_rds dont work i have made all what is listed including the make  but the program doesnt work

StoneSet, 2018-12-30:
Do /home/pi//home/pi/PiFmRds/src/pi_fm_rds

DylanHP, 2018-12-30:
Thanks StoneSet I have another answer why I can only transmit for a few seconds like 30 sec my music over a determined frequency?? After this 30 seconds the radio transmission restart

StoneSet, 2018-12-30:
Which command do you use?

saintofinternet, 2019-01-15:
> 
> 
> help me the command sudo ./pi_fm_rds dont work i have made all what is listed including the make but the program doesnt work

i am still getting the same error and it's not getting solved as per the instructions on this page. can someone help?

sarimbinwaseem, 2019-01-26:
Some problem ... nothing works

ChristopheJacquet, 2019-07-31:
This is most likely a duplicate of #126 which is fixed now (see #125). I also updated the README to explain that the pi_fm_rds executable is missing iff make fails.

Marking this issue as fixed (because #126 was fixed). Please open a new issue with the error message if make is failing for any other reason.

closed 2019-07-31 by ChristopheJacquet