ChristopheJacquet/PiFmRds #119: "make" not compiling correctly

SEPROG, 2018-12-25:
Hi, as stated in the title the make command is not compiling the way it should. After compiling the "pi_fm_rds" file doesn't exist. While compiling following errors occured:

rds.c: In function 'set_rds_rt':
rds.c:240:5: warning: 'strncpy' specified bound 64 equals destination size [-Wstringop-truncation]
     strncpy(rds_params.rt, rt, 64);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rds.c: In function 'set_rds_ps':
rds.c:247:5: warning: 'strncpy' specified bound 8 equals destination size [-Wstringop-truncation]
     strncpy(rds_params.ps, ps, 8);

"uname -a" gives me following
Linux kali 4.14.71-v8 #1 SMP PREEMPT Wed Oct 31 21:41:06 UTC 2018 aarch64 GNU/Linux

saintofinternet, 2019-01-15:
same error.... screenshot attached...
[image]

yoerithor, 2019-03-30:
Same, any solutions?

ChristopheJacquet, 2019-07-31:
The text you pasted is just an (innocuous) warning. The problem was that the architecture name (result of uname -m) was not recognized as a RPi. So `make` did not generate `pi_fm_rds`.

This should now be fixed (#117).

closed 2019-07-31 by ChristopheJacquet