ChristopheJacquet/PiFmRds #123: WAV Plays in Slow Motion

mcgregol, 2019-02-14:
Whenever I run it it works but it is in slow mo.  Does anybody know how to fix it?

Fanny1203, 2019-05-07:
Hello, I have the same problem.
Thank you for sharing this great project!

xbh, 2019-05-12:
Try tunning the ppm tp10000000

mcgregol, 2019-05-12:
@xbh Ok, sounds good.

Fanny1203, 2019-05-16:
Thank you! Actually, it worked fine without changing the parameter today.

preetjdp, 2019-06-12:
Thanks that worked

> Try tunning the ppm to10000000

ChristopheJacquet, 2019-08-01:
You shouldn't have to tinker with frequency correction. Marking as a duplicate of #75. Please comment there with your setup (distribution, potential usage of a GUI).

closed 2019-08-01 by ChristopheJacquet

ChristopheJacquet, 2019-08-01:
I found a recommendation to add gpu_freq=250 to /boot/config.txt on the RPi3. Source and explanations: https://github.com/SaucySoliton/PiFmRds/commit/41c6523fc65894a25efa3e1b8d921201679da7ca

Does this work for you?

Petrkiller2, 2024-02-13:
> Try tunning the ppm to10000000

A milion (1000000) worked instead of 10 milion.
I have RPI 3B+.