ChristopheJacquet/PiFmRds #126: Make issue

iocohen, 2019-03-23:
root@kali-pi:~/PiFmRds/src# make
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 rds.c
rds.c: In function ‘set_rds_rt’:
rds.c:240:5: warning: ‘strncpy’ specified bound 64 equals destination size [-Wstringop-truncation]
     strncpy(rds_params.rt, rt, 64);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rds.c: In function ‘set_rds_ps’:
rds.c:247:5: warning: ‘strncpy’ specified bound 8 equals destination size [-Wstringop-truncation]
     strncpy(rds_params.ps, ps, 8);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 waveforms.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 pi_fm_rds.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 fm_mpx.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 control_pipe.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 mailbox.c
mailbox.c: In function ‘mbox_open’:
mailbox.c:258:52: warning: implicit declaration of function ‘makedev’ [-Wimplicit-function-declaration]
     if(mknod(LOCAL_DEVICE_FILE_NAME, S_IFCHR|0600, makedev(MAJOR_NUM_A, 0)) >= 0 &&
                                                    ^~~~~~~
gcc -o pi_fm_rds rds.o waveforms.o mailbox.o pi_fm_rds.o fm_mpx.o control_pipe.o -lm -lsndfile
/usr/bin/ld: mailbox.o: in function `mbox_open':
/root/PiFmRds/src/mailbox.c:258: undefined reference to `makedev'
/usr/bin/ld: /root/PiFmRds/src/mailbox.c:265: undefined reference to `makedev'
collect2: error: ld returned 1 exit status
make: *** [Makefile:23: app] Error 1

basimci, 2019-03-27:
issue with kali on pi

try with raspberry os it will work

iocohen, 2019-03-27:
with rapsberypi os i got 
```
pi@raspberrypi:~/Desktop/PiFmRds/src $ make
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 pi_fm_rds.c
pi_fm_rds.c:105:21: fatal error: sndfile.h: No such file or directory
 #include <sndfile.h>
                     ^
compilation terminated.
Makefile:44: recipe for target 'pi_fm_rds.o' failed
make: *** [pi_fm_rds.o] Error 1
```

iocohen, 2019-03-27:
sorry had to install `sudo apt-get install libsndfile1-dev`. now is all fine is working on raspbian.

ChristopheJacquet, 2019-07-31:
Pull request #125 fixed the "makedev missing" error. Closing.

closed 2019-07-31 by ChristopheJacquet