ChristopheJacquet/PiFmRds #127: ./pi_fm_rds  is missing

XenonXDX123, 2019-04-02:
Well, I don't know why but when I try typing in sudo ./pi_fm_rds -audio sound.wav, I get the error that this file doesn't exist. I did run all the command correctly, I think... 
When I look into the src folder of the project on github I can actually see, that its missing there too.
What shall I do?

hxxr, 2019-04-22:
Did you compile the program?
```bash
git clone https://github.com/ChristopheJacquet/PiFmRds.git
cd PiFmRds/src
make clean
make
```
\
Did you `cd` into the directory to run the program?
```bash
cd ~/PiFmRds/src    # Or wherever the src directory is
sudo ./pi_fm_rds -audio sound.wav
```

bpowter, 2019-04-23:
I had this problem and had to edit the source code on mailbox.c. Click on the pull release tab and the solution is there under Fix compile error on Buster. Now sudo ./pi_fm_rds -audio sound.wav works for me.

awp8883, 2019-05-24:
I have this same problem

CodeDev1809, 2019-07-24:
Has anyone found a solution to this problem? I am currently dealing with the same error.

ChristopheJacquet, 2019-07-31:
This is most likely a duplicate of #126 which is fixed now (see #125). I also updated the README to explain that the `pi_fm_rds` executable is missing iff `make` fails.

Marking this issue as fixed (because #126 was fixed). Please open a new issue with the error message if `make` is failing for any other reason.

closed 2019-07-31 by ChristopheJacquet