ChristopheJacquet/PiFmRds #128: Changing audio input during broadcast

armartar, 2019-05-16:
Hi, 

for a project, I would need to switch between two audio inputs. Is this possible without stopping a broadcast or outputting to FIFO? If I start the first broadcast with audio input A and second broadcast with audio input B stopping first broadcast process stops both broadcasts. Is there any way to stop broadcast A in the background without stopping broadcast B?

btweb, 2019-05-16:
I'm not sure I fully understand your requirement, but you might want to look at liquidsoap (https://www.liquidsoap.info/). I found it has a fairly steep learning curve (it's extremely configurable), but it is very good for switching audio on the fly. I use it to fail over from a failed stream source (and back again). It can probably output to a pipe, or you could do it all in streams and pipe the stream to the broadcast process.

armartar, 2019-05-17:
Hi,

thanks for the quick response. Do you have any demo code available which works directly with PiFmRds?

ChristopheJacquet, 2024-03-02:
I would surely be possible to add this feature, but I feel it's beyond the scope of an experimental program like Pi-FM-RDS. In true Unix philosophy, you should look for (or build!) a program to switch between sources, and then pipe its output into Pi-FM-RDS, as suggested by [btweb](https://github.com/btweb) above.

closed 2024-03-02 by ChristopheJacquet