ChristopheJacquet/PiFmRds #129: sudo: ./pi_fm_rds: command not found

awp8883, 2019-05-24:
I clone, and do make clean and make in the src dir, and I try sudo: ./pi_fm_rds and it says command not found, I do ls and the pi_fm_rds file isnt even there.

Contents of my src folder:

Makefile        fm_mpx.h               mailbox.c        pulses.wav  rds_wav    sound_22050.wav   waveforms.o
control_pipe.c  fm_mpx.o               mailbox.h        rds.c       rds_wav.c  stereo_44100.wav
control_pipe.h  generate_pulses.py     noise_22050.wav  rds.h       rds_wav.o  waveforms.c
fm_mpx.c        generate_waveforms.py  pi_fm_rds.c      rds.o       sound.wav  waveforms.h

colivier74, 2019-06-02:
try `sudo ./pi_m_rds`

awp8883, 2019-06-02:
Tried. Didn’t work

ChristopheJacquet, 2019-07-31:
This is most likely a duplicate of #126 which is fixed now (see #125). I also updated the README to explain that the `pi_fm_rds` executable is missing iff `make` fails.

Marking this issue as fixed (because #126 was fixed). Please open a new issue with the error message if `make` is failing for any other reason.

closed 2019-07-31 by ChristopheJacquet