ChristopheJacquet/PiFmRds #134: PiFmRds on Raspberry Pi 4

Salisito, 2019-10-04:
Hi
First of all sorry for my English.
I've buyed a Raspberry Pi 4, 2gb RAM. Before I've had the Raspberry Pi 3 with PiFmRds working. Now I've installed PiFmRds on the raspi 4 and when I run the command, the console says me that it's working in X frequency but in really not, when I tune the frequency that I've set, there's white noise.
Has the new raspberry the gpio4 FM disabled or something like this?

Thanks

ChristopheJacquet, 2019-10-04:
Did you use the latest version of the code? Support for the Raspberry Pi 4 was contributed by @markondej a few days ago. I personally don't have a RPi 4 to test.

Salisito, 2019-10-04:
I've installed PiFmRds just yesterday using git clone https://github.com/ChristopheJacquet/PiFmRds.git

markondej, 2019-10-04:
I've just double checked it on RPi4 2GB and on July's Raspbian lite distro it works like a charm (I do always use lite distros). Maybe there's a problem with DMA channels, try changing definition of DMA_BASE_OFFSET  from 0x00007000 to 0x00007500, 0x00007900 etc. (there are 15 DMA channels available).
I'll try to test it on newest Raspbian (released week ago) and will let you know if any issues occur.

Salisito, 2019-10-04:
I would install lite distro and I'll try again. At the moment I've changed DMA channels and it stills not work.

Salisito, 2019-10-04:
I've installed Raspi Lite and PiFmRds and still don't work. When I play something, there are like some interferences. I've also tryied in other frequencies. I attach video.

https://drive.google.com/open?id=1CMSGVK1XaeZNwmoAK21N4mxBpRP288cy

Salisito, 2019-10-04:
@markondej I've also tryied to install your repository and it doesn't work (https://github.com/markondej/fm_transmitter)

markondej, 2019-10-06:
Checked newest Raspbian release and yes, it doesn't work (26 sep 2019). When simply simulating carrier via clock output (on GPIO4) the generated signal is noisy. I have no idea if this is problem with output itself or with clock source (PLLD). Like I wrote earlier everything seemed to work on July's release. I guess they messed something up with hardware (were there any GPU performance releated changes etc?). I will try to change clock source to any other and find out if this would work.

Salisito, 2019-10-06:
Okey, I hope you can fix it!!

Thaanks

Salisito, 2019-10-11:
@markondej have you been able to solve the problem?

markondej, 2019-10-13:
Nope, but I've managed to bypass it a little bit. You can switch to GPIO21 instead of GPIO4. I made branch with this "feature".

https://github.com/markondej/PiFmRds

Salisito, 2019-10-14:
It doesn't work for me...

captroy, 2019-11-01:
I am also having an issue transmitting with the PI 4. I pulled the branch using GPIO21, and also had could not get this working properly. 

I can see the change in GQRX when I start he fm_rds command, but, nothing is distinguishable.

TRSO4, 2019-12-28:
I seem to be having the same issue =/

waterwin, 2020-01-13:
Maybe this is a pointer to the right solution

Richard G Hirst added RPi4B support to his ServoBlaster and comments on DMA as follows:

The DMA channel used depends on the SOC type. If the Pi has a BCM2711
(Pi4B only, at present) then it uses channel 7; otherwise it uses channel 14.

Source = https://github.com/richardghirst/PiBits/tree/master/ServoBlaster

henry-shadowjet, 2020-02-03:
Hi all,
Thanks for the very useful and neat program.

I have the Raspberry Pi4 with 4GB RAM, running the latest Raspbian Buster distro. Downloaded and compiled the source with no issues (thank you!).

However the problem is I can't seem to get the signal to output an audible signal. The progeam does not generate any error, but when I tune my car radio to the signal it's supposedly outputting signal to, the only thing I can hear is pulses. Something is there though because it is clearly different when the program is running vs not running. But yeah, all I can hear is small pulses, no matter which .wav I used on the -audio tag (I tried all the different supplied samples).

Any pointer in the right direction would be much appreciated! I know a problem is already recognised in the above convo, just in case this is a bit different.

Thanks in advance, and keep up the great work!

JokrCode, 2020-02-04:
> Hi all,
> Thanks for the very useful and neat program.
>
> I have the Raspberry Pi4 with 4GB RAM, running the latest Raspbian Buster distro. Downloaded and compiled the source with no issues (thank [...]

I'm in the same boat. I have rasberry pi 4 2gigs. I'm running kali and ssh into it I feel like everything's install commands run but nothing happens. Just s tiny bit of static? I'm not sure if it's running properly. Maybe my signal isnt strong enough but the cord is right next to my antenna. I think this is updated to work for pi 4 but idk if the make clean and make command is working? Let me know if you get it to work.

henry-shadowjet, 2020-02-06:
There was just a kernel update today to 4.19.97-v7l+, so I tried it again with much anticipation, but unfortunately yielded the same result. Still the same static and pulses.

I do have a short wire same as you do, but the radio antenna and the wire is pretty much touching each other so I assume that's not an issue, but thought I should mention it just in case.

Thanks everyone. Any info would be appreciated.

JokrCode, 2020-02-15:
Same with mine I get nothing but I do see something happening but seems not strong enough to take over. Theres another kernel update recently but I haven't done either of them. Is there a way to do it on kali I just dont wanna reinstall it all again just to update the kernel

henry-shadowjet, 2020-02-16:
Since the new kernel update, I can't even get the pulsing sound I get before. Now it's as if nothing happens when I activate the program. Go figure. Haha

markondej, 2020-03-21:
Lately I've had some time play with spectrum analyzer, and this is what I found to "work".
First: try switching antena to GPIO21 (last PIN on RPi4 header) using my branch ( https://github.com/markondej/PiFmRds ). Before you start transmitting switch ARM core frequency scaling governor to "performance" settings:
```
echo "performance"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
```
Finally use frequencies below 93.5 MHz
```
sudo ./pi_fm_rds -audio pulses.wav -freq 92
```

Salisito, 2020-03-21:
Wow @markondej, now it works!! But the audio quality is not like before...

Salisito, 2020-03-22:
There's like a distorsion

goutx2, 2020-04-21:
Hello
I am trying to get transmission above 93 mhz but ni chance 
My question is why we cant transmite above 93 mhz 
And i want tell you the published "bcm2711 arm periferals.pdf" 
 take a look at 
https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bcm2711

Budgiebrain994, 2020-04-26:
Hi there - same deal. I'm on a Raspberry Pi 4, with 4GB of RAM. But instead, I've installed Ubuntu 64bit. I've had no luck running the application, or any available variation. It uses a different architecture (armv8a + aarch64). Given the FPU is not available on this platform, am I right in speculating we won't be having much luck for a while? I'm not good with such low-level code myself.

makeworld-the-better-one, 2020-05-11:
The Pi 4 is known to have issues, see [this repo](https://github.com/markondej/fm_transmitter#raspberry-pi-4) that mentions it.

doctorsax, 2020-05-23:
Bonjour, quand je veux compilé le programme sur ma raspberry pi 4 4g en utilisant make, il me met des message d'erreur comment les résoudre ?

ChristopheJacquet, 2020-07-08:
@doctorsax A priori le programme ne fonctionne pas, ou pas bien, sur RPi 4. Cependant la compilation devrait réussir. Vérifiez que vous avez bien toutes les dépendances listées dans le README.

doctorsax, 2020-07-09:
@ChristopheJacquet Bonjour j'ai regarder mais toujours le même message d'erreur
 [image]

makeworld-the-better-one, 2020-07-09:
@doctorsax C'est pas une erreur, mais juste un avertissement. Tu peux l'ignorer, je pense.

doctorsax, 2020-07-09:
@makeworld-the-better-one Justement j'ai essayer mais le programme ne veut pas se lancer

makeworld-the-better-one, 2020-07-09:
Quel est le résultat quand tu exécutes `./pi_fm_rds`? Je ne suis pas le developer, mais je sais que c'est cette résultat que Christophe a besoin.

Ben1701A, 2021-01-20:
Weird. I'm running the compiled binary just fine on my Raspberry Pi 400.

captroy, 2021-09-20:
Hey all, I think I have a fix for this issue. I've successfully transmitted RDS on my pi 4, and put a pull request for the code change.  Fingers crossed this works for group!

Ben1701A, 2021-09-20:
> Hey all, I think I have a fix for this issue. I've successfully transmitted RDS on my pi 4, and put a pull request for the code change. Fingers crossed this works for group!
> 
> I should mention, this is thanks to @1anakin20 for his fix on the pi-rc project. @ @

Ooooo. Can't wait to test once I get back home.

jurosic, 2021-10-29:
still cant get this to work on my rpi4 running raspbian lite, i have an antenna and i am standing next to mi radio but only white noise, there is a difference when i start/stop the broadcast, but no actual audio plays

StefCoders, 2023-08-22:
This worked perfectly `echo "performance"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor`

comicsansgreenki, 2023-08-22:
@StefCoders ooh nice! did this let you use a display output with the pi4 simultaneously?

StefCoders, 2023-08-22:
Nope, headless pi.

Salisito, 2023-08-23:
> This worked perfectly `echo "performance"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor`

@StefCoders what else did you do to make work PiFmRds? Which repository did you use? Thanks in advance

StefCoders, 2023-08-23:
I just used this one. Nothing else.

crisdash96, 2024-07-10:
this Soultion witch not working pifmrds fm transmitter

Put the cable in pin Numbers 6

Salisito, 2024-10-18:
Hi all, 
It's been a long time since this issue was reported. Has someone achieved to make work PiFmRds in a RPI4?

Samgames333, 2024-12-10:
I have used a similar program (PiFmAdv) and it works on both of my 4s. I am not sure why it works so much better but the noisy audio, limit at 93.5mhz, and muffled audio are all fixed. There are some benefits with PiFmRds as the stereo pilot reaches -20dBFS instead of what I am guessing is around -22 with Adv (small stuff here) and the way it deactivates the signal as PiFmAdv will continue transmitting a bare carrier instead of stopping the transmission, which PiFmRds does well. Both seem to have a similar modulation efficiency with a flat area at the top that is about half the size of a commercial transmitter. The audio quality of Adv is better but the volume by default is very low, but I found out that this can be fixed by setting the --mpx parameter to >50. The RDS works well on both but I had an issue with the control pipe in PiFmRds, but that could just be because my python script to change ps text and rt to display the song name was made and tested for PiFmAdv. I have great respect for both options, but if you are using a Pi 4, PiFmAdv is the way to go for now. Hope this helped.

PiFmRds:
[image]
[image]

PiFmAdv:
[image]
[image]

leojbgiusti, 2026-01-18:
Not working for myself on a Pi4. It compiles and runs but no output. PiFmAdv however won't even compile on my Pi4.