ChristopheJacquet/PiFmRds #135: Compiloing for raspberry zero W

ricksmith96, 2019-12-28:
When you go to src/ and make clean then make its spits out error to fix the error you need to edit "-mfloat-abi=hard " to "-mfloat-abi=soft"

Dtpk, 2020-02-19:
Came here to say setting that parameter to " -mfloat-abi=softfp " works on Kali Linux distro's quite well. If you get audio skipping after using just "soft" set it too softfp this should allow it to play back smooth and compile.

ChristopheJacquet, 2025-11-30:
Currently, hard float ABI works fine with Raspberry Pi OS. Made a note in README to mention different setting needed for other distributions.

https://github.com/ChristopheJacquet/PiFmRds?tab=readme-ov-file#compiling-on-distributions-with-different-float-abis

Closing issue.