ChristopheJacquet/PiFmRds #136: AUX audio pipe error

mishailovic, 2020-01-25:
when i am trying to pipe the AUX input of a sound card into Pi-FM-RDS with sudo arecord -fS16_LE -r 44100 -Dplughw:1,0 -c 2 -  | sudo ./pi_fm_rds -audio -
I get this error:
``` Using mbox device /dev/vcio.                                
Allocating physical memory: size = 3403776     mem_ref = 5     bus_addr
= fe7b0000     virt_addr = 0x767f7000                       
ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
arecord: main:828: audio open error: No such file or directory
ppm corr is 0.0000, divider is 1096.4912 (1096 + 2012*2^-12) [nominal 10
96.4912].                                                   
Error: could not open stdin for audio input.                
Terminating: cleanly deactivated the DMA engine and killed the carrie ```

How to fix it?

mishailovic, 2020-01-25:
Yeah, and i am do this on my Raspberry Pi 3b+

olegoleg1111121, 2021-01-30:
I have a this problem. Help please!(

mishailovic, 2021-01-30:
I think this issue happening because author using rpi 2, and rpi 3 has different sound board

olegoleg1111121, 2021-02-01:
I fixed it, if you want to use it with a mic, plug it in and set ossb (for audio to work) it might be called differently

mishailovic, 2021-02-01:
> Raspberry cannot use mic in aux. Only usb!!!!
So i will close an issue, thanks

closed 2021-02-01 by mishailovic