ChristopheJacquet/PiFmRds #137: Playing Spotify on PiFmRds?

wink05, 2020-01-26:
Hi! I'm looking for someone who may help me with my little idea.
I have an idea to stream music from Spotify to FM. I thought about it and I think its able to do using just a sound from sound card, but I don't have idea how to do make Spotify play on Raspberry Pi. 
Maybe there is somebody who did this or want to help me?
any help will be useful!

jurosic, 2021-10-30:
Sorry for the really late response, but this should be possible, tho you will need to use another repo, not pifmrds, look up bt-speaker for raspberry pi, it turns Bluetooth input into AUX output, and you could look into he code and make it redirect audio to something like pfmrds or you could use piping "|" or use some audio controllers for linux like pulseaudio

jurosic, 2021-10-30:
For example this could be possible with piping audio, so pipe audio from a capture card (bluetooth hci0) and feed it to pifmrds, but i dont know how to do this, let me know if you manage to do anything

ChristopheJacquet, 2024-03-02:
That's an interesting idea, but I'm going to close this as out of scope for the project. Pi-FM-RDS can transmits a sound file as FM+RDS. Converting other sources as a sound stream that can be piped into Pi-FM-RDS (or any other program) is a different project.

closed 2024-03-02 by ChristopheJacquet