ChristopheJacquet/PiFmRds #141: Very high pitch noise while broadcasting

scratchermatch, 2020-03-05:
Whenever I broadcast, I will get the song just fine, but I also get a very high pitch sound in the background that really hurts my ears.
I can mitigate the problem by broadcasting at really low frequencies, but it's still there, faintly audible, and killing my ears.

Am I missing something stupid? How could I get rid of the noise?
I sped each of my audio clips up to x2 speed so that they played normally, which I hear increases the pitch, but even the slow one had the high pitch noise in the background.

profiluefter, 2020-04-24:
I have the same problem but only after already playing something a few times

evertverduin, 2020-12-30:
This relates to my problem as well!