ChristopheJacquet/PiFmRds #143: STREAM AND BROADCASTING

sanustudio-dev, 2020-04-02:
Hello,
At this time i need your help about this PiFmRds, 
1st question if i have a licence and a square-wave filter can i use my PiFmRds ?

And 2nd question how can i play a webradio stream into PiFmRds
Thanks to advance.

Marrley, 2020-06-25:
not sure about the licence but here is for the stream:

sox -t mp3 http://myfancy.stream.url/mp3 -t wav - | pi_fm_rds -freq 106.0 -audio -

[https://github.com/ChristopheJacquet/PiFmRds/issues/102](https://github.com/ChristopheJacquet/PiFmRds/issues/102)

ChristopheJacquet, 2021-01-04:
About the licence: never emit radio waves if you want to be on the safe side. And if you believe you can, I can only recommend talking with a legal counsel before doing proceeding...

On streaming: did the answer above work for you?

closed 2024-03-02 by ChristopheJacquet